firesim/firesim #708: TracerV TriggerSource/TriggerSink annotation not working

bgottschall, 2021-01-28:
I created a new tracing interface which allows me to pipe out additional informations out of the simulation. I've created this interface exactly like TracerV, but a little bit simpler as it is just tracing out a bit vector. The interface works exactly as expected and I retrieve so far in the Verilator simulation exactly the data from my Bridge that I want.
Now I wanted to reuse the TracerV trigger to enable tracing between certain trigger points. The documentation talks about the TriggerSource and TriggerSink annotation and the trigger is annotated as far as I can see like that in the TracerV Bridge.

So I tried to do the following:
```scala
object GenericTraceBridge {
  def apply(generic_trace: TileGenericTraceIO)(implicit p:Parameters): GenericTraceBridge = {
    val ep = Module(new GenericTraceBridge(generic_trace.bitWidth))
    withClockAndReset(generic_trace.clock, generic_trace.reset) { TriggerSink(ep.io.tracevTrigger, noSourceDefault = false.B) }
    ep.io.trace := generic_trace
    ep
  }
}
```

However it gives me during the FIRRTL transformation the following error:
```text
[info] running midas.stage.GoldenGateMain -o /home/bj/github/firesim/sim/generated-src/f1/FireSim-FireSimSmallBoomConfig-BaseF1Config/FPGATop.v -i /home/bj/github/firesim/sim/generated-src/f1/FireSim-FireSimSmallBoomConfig-BaseF1Config/firesim.firesim.FireSim.FireSimSmallBoomConfig.fir -td /home/bj/github/firesim/sim/generated-src/f1/FireSim-FireSimSmallBoomConfig-BaseF1Config -faf /home/bj/github/firesim/sim/generated-src/f1/FireSim-FireSimSmallBoomConfig-BaseF1Config/firesim.firesim.FireSim.FireSimSmallBoomConfig.anno.json -ggcp firesim.firesim -ggcs BaseF1Config --no-dedup -E verilog
Elaborating design...
Done elaborating.
[error] (run-main-0) firrtl.CustomTransformException: 
[error] firrtl.CustomTransformException: 
[error] 	at firrtl.stage.transforms.CatchCustomTransformExceptions.execute(CatchCustomTransformExceptions.scala:13)
[error] 	at firrtl.Transform.transform(Compiler.scala:311)
[error] 	at firrtl.Transform.transform$(Compiler.scala:311)
[error] 	at firrtl.stage.transforms.CatchCustomTransformExceptions.transform(CatchCustomTransformExceptions.scala:7)
[error] 	at firrtl.stage.transforms.UpdateAnnotations.$anonfun$internalTransform$1(UpdateAnnotations.scala:22)
[error] 	at firrtl.Utils$.time(Utils.scala:169)
[error] 	at firrtl.Transform$.runTransform(Compiler.scala:209)
[error] 	at firrtl.stage.transforms.UpdateAnnotations.internalTransform(UpdateAnnotations.scala:22)
[error] 	at firrtl.stage.transforms.UpdateAnnotations.internalTransform(UpdateAnnotations.scala:8)
[error] 	at firrtl.options.Translator.transform(Phase.scala:248)
[error] 	at firrtl.options.Translator.transform$(Phase.scala:248)
[error] 	at firrtl.stage.transforms.UpdateAnnotations.transform(UpdateAnnotations.scala:8)
[error] 	at firrtl.options.DependencyManager.$anonfun$transform$3(DependencyManager.scala:278)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error] 	at scala.collection.immutable.List.foldLeft(List.scala:89)
[error] 	at firrtl.options.DependencyManager.transform(DependencyManager.scala:269)
[error] 	at firrtl.options.DependencyManager.transform$(DependencyManager.scala:255)
[error] 	at firrtl.stage.TransformManager.transform(TransformManager.scala:14)
[error] 	at firrtl.stage.TransformManager.execute(TransformManager.scala:22)
[error] 	at midas.stage.GoldenGateCompilerPhase.transform(GoldenGateCompilerPhase.scala:40)
[error] 	at midas.stage.GoldenGateCompilerPhase.transform(GoldenGateCompilerPhase.scala:16)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error] 	at firrtl.options.Translator.transform(Phase.scala:248)
[error] 	at firrtl.options.Translator.transform$(Phase.scala:248)
[error] 	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error] 	at midas.stage.GoldenGateStage.$anonfun$run$1(GoldenGateStage.scala:29)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error] 	at scala.collection.immutable.List.foldLeft(List.scala:89)
[error] 	at midas.stage.GoldenGateStage.run(GoldenGateStage.scala:29)
[error] 	at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[error] 	at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error] 	at firrtl.options.Translator.transform(Phase.scala:248)
[error] 	at firrtl.options.Translator.transform$(Phase.scala:248)
[error] 	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error] 	at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:47)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error] 	at scala.collection.immutable.List.foldLeft(List.scala:89)
[error] 	at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:47)
[error] 	at logger.Logger$.$anonfun$makeScope$2(Logger.scala:166)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[error] 	at logger.Logger$.makeScope(Logger.scala:164)
[error] 	at firrtl.options.Stage.transform(Stage.scala:47)
[error] 	at firrtl.options.Stage.execute(Stage.scala:58)
[error] 	at firrtl.options.StageMain.main(Stage.scala:71)
[error] 	at midas.stage.GoldenGateMain.main(GoldenGateStage.scala)
[error] 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error] 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error] 	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] 	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
[error] Caused by: firrtl.passes.CheckHighFormLike$UndeclaredReferenceException: : [module FireSim] Reference gentrace_generic_traces_0_clock is not declared.
```

For reference here is the link to the mailing list discussion about that issue: [https://groups.google.com/g/firesim/c/YQHhnuGdHMI](https://groups.google.com/g/firesim/c/YQHhnuGdHMI)

**FireSim Version:** 1.11.0

I've also uploaded the FIRRTL and annotation file (links will be valid until issue is resolved)
Annotation JSON: [https://bjorngot.idi.ntnu.no/firesim/anno.json.xz](https://bjorngot.idi.ntnu.no/firesim/anno.json.xz)
FIRRTL: [https://bjorngot.idi.ntnu.no/firesim/smallboom.fir.xz](https://bjorngot.idi.ntnu.no/firesim/smallboom.fir.xz)

davidbiancolin, 2021-01-30:
Thanks Bjorn, i'm looking into it.

davidbiancolin, 2021-06-03:
@bgottschall, i'm sorry i never got around to fixing this. 

The links are dead now -- did you workaround the problem?

bgottschall, 2021-06-03:
For my purpose I did work around this problem by just not using any triggers and tracing every cycle. I just linked my target application into a address range that was unique so I could identify the data of interest.

But it would be great to attach to the tracerV trigger to selectively trace. I give you new links, my university decided to put this server behind a firewall ;)
[https://bgottschall.de/smallboom.fir.xz](https://bgottschall.de/smallboom.fir.xz)
[https://bgottschall.de/anno.json.xz](https://bgottschall.de/anno.json.xz)