firesim/firesim #713: Enhance launchrunfarm to patiently try harder

timsnyder-siv, 2021-02-04:
We have a workload that contains 60 jobs.  Haven't ever had a problem getting hosts but yesterday and today, we have started hitting this case:
https://github.com/firesim/firesim/blob/9ffba8dbf4ec22a17269ecd15f0a6ad761c8e82b/deploy/awstools/awstools.py#L174-L178

It's annoying because:
* we've already allocated a goodly portion of what we need
* `run_workload.sh` very politely does `terminaterunfarm` after `launchrunfarm` bombs out
* then we start trying to allocate again.

Seems to me like when AWS is having trouble meeting demand for f1 hosts that `launchrunfarm` should:
* look for existing hosts marked with `fsimcluster` and only try to allocate the remaining (so that it can be rerun after a failure)
* not give up completely after failing to allocate a host across all subnets.  I've locally hacked it so that it just tries forever right now.  It should probably be a config_runtime.ini parameter for how long we try to allocate hosts before giving up.

@davidbiancolin , how often do you guys run into this type of thing?  Do my changes to `launchrunfarm` seem sensible?  If so, I'm happy to open a PR.

davidbiancolin, 2021-02-04:
I have run into stuff like this in the past, but i don't tend to use workloads that are quite that large. 

Both looking for existing hosts with the same runfarm tag (and asking for the remainder), and adding a timeout would be good.

timsnyder-siv, 2021-02-04:
Cool.  I added a runtime_config.ini parameter for the timeout and plumbed it all up.  Probably needs a test but I'm going to submit what I have so far so that it doesn't get lost.

sagark, 2022-03-31:
@timsnyder-siv this is solved, right?

timsnyder-siv, 2022-04-01:
Yes.  We got it merged into 1.13.

closed 2022-04-01 by timsnyder-siv