firesim/firesim #714: Gemmini fails to meet timing requirements

spaceotter, 2021-02-04:
The build configuration firesim-rocket-singlecore-gemmini-no-nic-l2-llc4mb-ddr3 doesn't work:
```
[████] out: FATAL: Design did not meet timing requirements. Terminating.
[████] out: INFO: [Common 17-206] Exiting Vivado at Thu Feb  4 06:14:47 2021...
[████] out: 
Warning: run() received nonzero return code 3 while executing './aws_build_dcp_from_cl.sh -foreground'!
FireSim FPGA Build Failed
Your FPGA build failed for triplet: FireSim-DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.GemminiRocketConfig-F110MHz_BaseF1Config.
Inspect the log output from IP address ████ for more information.
Terminating the build instance now.
```
Are there adjustments that have to be made beyond selecting it in config_build.ini?
I'm going to try again with a lower clock in PLATFORM_CONFIG.

davidbiancolin, 2021-02-04:
Using a slower clock in PLATFORM_CONFIG is the right solution here. @alonamid, can you update the gemmini recipes to be a little more conservative?

spaceotter, 2021-02-05:
So far it failed at 110MHz and 90Mhz.

alonamid, 2021-02-05:
I'm currently testing this as well. I'm currently waiting for the 70 MHz build attempt

spaceotter, 2021-02-06:
I just got a failure at 65MHz. Does it do a clean build every time? Are my changes to  config_build_recipes.ini actually taking effect? The target frequency is listed in the triplet. Do I need to drop the frequency even more or might there be an issue with the design?

alonamid, 2021-02-06:
I can confirm that it builds at 50 MHz. However, if you would like it to run at a higher frequency, it is likely that the timing failures are due to a recent change in Gemmini which added floating point scaling. By changing a couple of Gemmini parameters, Gemmini can be better pipelined to address these issues. The default Gemmini parameters will be updated soon to reflect this pipelining, but if you would like to change them manually, you can change the values of the following parameters to 4:
- [`mem_pipeline`](https://github.com/ucb-bar/gemmini/blob/e6e14f711760b976d8eb00c32d0fe2423aeda211/src/main/scala/gemmini/Configs.scala#L58) , [mvin_scale_args pipelining argument](https://github.com/ucb-bar/gemmini/blob/e6e14f711760b976d8eb00c32d0fe2423aeda211/src/main/scala/gemmini/Configs.scala#L125) (from 0 to 4), and the [acc_scale_args pipelining argument](https://github.com/ucb-bar/gemmini/blob/e6e14f711760b976d8eb00c32d0fe2423aeda211/src/main/scala/gemmini/Configs.scala#L168) (from 0 to 4).

spaceotter, 2021-02-07:
The last change doesn't work:
```
[localhost] out: [error] java.lang.AssertionError: assertion failed: Accumulator's scale latency must be 0 cycles
[localhost] out: [error]        ...
[localhost] out: [error]        at gemmini.GemminiArrayConfig.generateHeader(GemminiConfigs.scala:224)
```
I'll try the other two.

hngenc, 2021-02-07:
The `acc_scale_args` argument, as a special case, does not allow pipelining arguments above 0. To pipeline the acc-scaling path, you should instead increase `mem_pipeline` to 4.

hngenc, 2021-02-08:
I've just merged 2 PRs into `gemmini/dev` that should make it easier to reach timing:
* https://github.com/ucb-bar/gemmini/pull/55
* https://github.com/ucb-bar/gemmini/pull/56

spaceotter, 2021-02-08:
I got a different error with the gemmini/dev branch. I'm just going back to the original at 50MHz which alonamid said was working.
<details>

```
[████] out: INFO: [Synth 8-5816] Retiming module `bd_bf3f_ilmb_0__xdcDup__2`
[████] out:    Effective logic levels is 0, the effective logic levels of whole design is 71
[████] out:    Skipping retiming for this module
[████] out: 
[████] out: INFO: [Synth 8-5816] Retiming module `bd_bf3f_ilmb_0__xdcDup__2' done
[████] out: 
[████] out: 
[████] out: INFO: [Common 17-14] Message 'Synth 8-5816' appears 100 times and further instances of the messages will be disabled. Use the Tcl command set_msg_config to change the current settings.
[████] out: INFO: [Synth 8-6153] Backward retiming for flip-flop spadi_18/firesim_top/top/sim/target/FireSim/lazyModule/system/tile_prci_domain/tile_reset_domain_tile/gemmini/spad/writer/req_data_reg[72] along instance spadi_18/i_7090 fails due to incompatible flip-flops 
[████] out: 
[████] out: INFO: [Synth 8-6153] Backward retiming for flip-flop spadi_18/firesim_top/top/sim/target/FireSim/lazyModule/system/tile_prci_domain/tile_reset_domain_tile/gemmini/spad/writer/req_data_reg[73] along instance spadi_18/i_7088 fails due to incompatible flip-flops 
[████] out: 
[████] out: INFO: [Synth 8-6153] Backward retiming for flip-flop spadi_18/firesim_top/top/sim/target/FireSim/lazyModule/system/tile_prci_domain/tile_reset_domain_tile/gemmini/spad/writer/req_data_reg[74] along instance spadi_18/i_7086 fails due to incompatible flip-flops 
[████] out: 
[████] out: INFO: [Synth 8-6153] Backward retiming for flip-flop spadi_18/firesim_top/top/sim/target/FireSim/lazyModule/system/tile_prci_domain/tile_reset_domain_tile/gemmini/spad/writer/req_data_reg[75] along instance spadi_18/i_7084 fails due to incompatible flip-flops 
[████] out: 
[████] out: INFO: [Synth 8-6153] Backward retiming for flip-flop spadi_18/firesim_top/top/sim/target/FireSim/lazyModule/system/tile_prci_domain/tile_reset_domain_tile/gemmini/spad/writer/req_data_reg[76] along instance spadi_18/i_7082 fails due to incompatible flip-flops 
[████] out: 
[████] out: INFO: [Synth 8-6153] Backward retiming for flip-flop spadi_18/firesim_top/top/sim/target/FireSim/lazyModule/system/tile_prci_domain/tile_reset_domain_tile/gemmini/spad/writer/req_data_reg[77] along instance spadi_18/i_7080 fails due to incompatible flip-flops 
[████] out: 
[████] out: INFO: [Synth 8-6153] Backward retiming for flip-flop spadi_18/firesim_top/top/sim/target/FireSim/lazyModule/system/tile_prci_domain/tile_reset_domain_tile/gemmini/spad/writer/req_data_reg[79] along instance spadi_18/i_7092 fails due to incompatible flip-flops 
[████] out: 
[████] out: Abnormal program termination (11)
[████] out: Please check '/home/centos/firesim-build/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/hs_err_pid12884.log' for details
[████] out: 
Warning: run() received nonzero return code 139 while executing './aws_build_dcp_from_cl.sh -foreground'!
FireSim FPGA Build Failed
Your FPGA build failed for triplet: FireSim-DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.GemminiRocketConfig-F75MHz_BaseF1Config.
```

</details>

closed 2021-05-05 by timsnyder-siv