firesim/firesim #720: machine-launch-script.sh failing on CircleCI

timsnyder-siv, 2021-02-09:
The past few CI runs have failed during the `machine-launch-script.sh` with this message:
```
Collecting cryptography>=2.5 (from paramiko<3.0,>=1.10->fabric==1.14.0)
  Downloading https://files.pythonhosted.org/packages/35/52/a3b9c3d8ce84544bfe8d663ba993e0593d9c518d6c08f01f6f8fff87b895/cryptography-3.4.2.tar.gz (544kB)
    Complete output from command python setup.py egg_info:
    
            =============================DEBUG ASSISTANCE==========================
            If you are seeing an error here please try the following to
            successfully install cryptography:
    
            Upgrade to the latest pip and try again. This will fix errors for most
            users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
            =============================DEBUG ASSISTANCE==========================
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-CsWiDU/cryptography/setup.py", line 14, in <module>
        from setuptools_rust import RustExtension
    ImportError: No module named setuptools_rust
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-CsWiDU/cryptography/
You are using pip version 8.1.2, however version 21.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

Exited with code exit status 1

CircleCI received exit code 1
```

@davidbiancolin , have you seen this before?  Looks like it started failing yesterday on your 'pdes' branch and it also failed this morning for #719

davidbiancolin, 2021-02-09:
I have not, but it might be broken on AWS instances too and thus, master-breaking for new users...

davidbiancolin, 2021-02-09:
Seems like we might need to upgrade setup tools. @timsnyder-siv, are you going to look into this?

timsnyder-siv, 2021-02-09:
I was able to run https://github.com/firesim/firesim/blob/9ffba8dbf4ec22a17269ecd15f0a6ad761c8e82b/scripts/machine-launch-script.sh on a fresh c5.4xlarge running the 1.6.1 FPGA Developer AMI without hitting the issue that CircleCI seems to be having.

davidbiancolin, 2021-02-09:
Alright. I'll look at the CI later today

closed 2021-02-11 by timsnyder-siv