firesim/firesim #721: Running baremetal tests with esp-tools

spaceotter, 2021-02-09:
I'm trying to run gemmini tests with the gemmini rocket config. It's not clear how to do this. Here's what I tried:

- firesim is using riscv-tools, not esp-tools, which afaik is needed to emit the special instructions.
- so go to chipyard and back up env.sh, then `scl enable devtoolset-8 bash` `./scripts/build-toolchains.sh esp-tools`
- now, copy the gemmini aligned.c into the firesim-software/test/bare/hello.c and copy the gemmini-rocc-tests/include and gemmini-rocc-tests/rocc-software/src/xcustom.h to the firesim-software/test/bare
- with the new env.sh in chipyard sourced, marshal will build the new hello binary

However, trying to do infrasetup creates an error:
<details>

```
In file included from /home/centos/firesim/sim/firesim-lib/src/main/cc/bridges/tracerv/tracerv_dwarf.cc:1:0:
/home/centos/firesim/sim/firesim-lib/src/main/cc/bridges/tracerv/tracerv_dwarf.h:9:22: fatal error: libdwarf.h: No such file or directory
 #include <libdwarf.h>
                      ^
compilation terminated.
In file included from /home/centos/firesim/sim/firesim-lib/src/main/cc/bridges/tracerv/tracerv_elf.h:7:0,
                 from /home/centos/firesim/sim/firesim-lib/src/main/cc/bridges/tracerv/tracerv_elf.cc:1:
/home/centos/firesim/sim/firesim-lib/src/main/cc/bridges/tracerv/tracerv_dwarf.h:9:22: fatal error: libdwarf.h: No such file or directory
 #include <libdwarf.h>
                      ^
compilation terminated.
In file included from /home/centos/firesim/sim/firesim-lib/src/main/cc/bridges/tracerv/tracerv_elf.h:7:0,
                 from /home/centos/firesim/sim/firesim-lib/src/main/cc/bridges/tracerv/tracerv_processing.cc:2:
/home/centos/firesim/sim/firesim-lib/src/main/cc/bridges/tracerv/tracerv_dwarf.h:9:22: fatal error: libdwarf.h: No such file or directory
 #include <libdwarf.h>
                      ^
compilation terminated.
make[1]: *** [/home/centos/firesim/sim/output/f1/FireSim-DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.GemminiRocketConfig-F50MHz_BaseF1Config/FireSim-f1] Error 1
make: *** [/home/centos/firesim/sim/output/f1/FireSim-DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.GemminiRocketConfig-F50MHz_BaseF1Config/FireSim-f1] Error 2
2021-02-09 22:33:26,613 [build_fpga_d] [INFO ]  FPGA software driver build failed. Exiting. See log for details.
2021-02-09 22:33:26,613 [build_fpga_d] [INFO ]  You can also re-run 'make DESIGN=FireSim TARGET_CONFIG=DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.GemminiRocketConfig PLATFORM_CONFIG=F50MHz_BaseF1Config f1' in the 'firesim/sim' directory to debug this error.
2021-02-09 22:33:26,614 [<module>    ] [ERROR]  Fatal error.
Traceback (most recent call last):
  File "/home/centos/firesim/deploy/firesim", line 334, in <module>
    main(args)
  File "/home/centos/firesim/deploy/firesim", line 282, in main
    globals()[args.task](simconf)
  File "/home/centos/firesim/deploy/firesim", line 95, in infrasetup
    runtime_conf.infrasetup()
  File "/home/centos/firesim/deploy/runtools/runtime_config.py", line 380, in infrasetup
    self.firesim_topology_with_passes.infrasetup_passes(use_mock_instances_for_testing)
  File "/home/centos/firesim/deploy/runtools/firesim_topology_with_passes.py", line 390, in infrasetup_passes
    self.pass_build_required_drivers()
  File "/home/centos/firesim/deploy/runtools/firesim_topology_with_passes.py", line 373, in pass_build_required_drivers
    server.get_server_hardware_config().build_fpga_driver()
  File "/home/centos/firesim/deploy/runtools/runtime_config.py", line 203, in build_fpga_driver
    exit(1)
  File "/usr/lib64/python2.7/site.py", line 364, in __call__
    raise SystemExit(code)
SystemExit: 1
```

</details>

Is there a better way to do this? Has someone run the gemmini tests with firesim before?

spaceotter, 2021-02-10:
I was able to work around the issue by swapping the riscv-tools env.sh file back in and running infrasetup again. The test program seems to have worked. However trying to runworkload a second time crashes. So I would still like to know if there is a less sketchy workflow for this and also where runworkload is supposed to work twice on the same runfarm.

a0u, 2021-02-10:
`scripts/firesim-setup.sh` must be re-run after switching toolchains.  Part of the FireSim setup procedure involves building and installing libelf and libdwarf into `$RISCV`.

a0u, 2021-02-10:
If you're not using FireSim within Chipyard (i.e., FireSim is the top-level repo), you can change the toolchain variant installed by FireSim here:

https://github.com/firesim/firesim/blob/7e32128bea5390ca528052dcdffb13656ee41162/build-setup-nolog.sh#L163

spaceotter, 2021-02-12:
Thanks, changing that and running the script again seems have worked.
I'm still seeing that `firesim runworkload` doesn't work twice in a row. Is that expected?
```
TraceRV 0: Tracing disabled, since +tracefile was not provided.
TraceRV 1: Tracing disabled, since +tracefile was not provided.
random min: 0x0, random max: 0xffffffffffffffff
TracerV: Trigger enabled from 0 to 18446744073709551615 cycles
TracerV: Trigger enabled from 0 to 18446744073709551615 cycles
Commencing simulation.
terminate called without an active exception

Script done on Fri 12 Feb 2021 09:45:15 AM UTC
```

jerryz123, 2021-02-12:
You generally need to run `firesim infrasetup` before each `firesim runworkload`.

spaceotter, 2021-02-13:
That was it. Thanks guys!

closed 2021-02-13 by spaceotter