firesim/firesim #729: TracerVBridge issue with HeteroConfig

SeahK, 2021-02-26:
When I tried to build a fpga image with HeteroConfig, I faced TracerV issue that "ports of extmodule TracerVBridge with defname TracerVBridge are different for an extmodule with the samde defname"
For now, I commented out TracerVBridge at BridgeBinder.scala, and it was built. (Though it wouldn't ahve autocounter features)
[image]

davidbiancolin, 2021-04-15:
Hey Seah can you provide the tuple to reproduce this.

SeahK, 2021-05-03:
Sorry for late response. The build config is this:

[firesim-rocket-heterocore30-16x16_2048-gemmini]
DESIGN=FireSim
TARGET_CONFIG=DDR3FRFCFS_FireSimGemminiRocketHeteroConfig
PLATFORM_CONFIG=MCRams_F30MHz_BaseF1Config
instancetype=z1d.2xlarge
deploytriplet=None

For the target config, I included
  new WithDefaultFireSimBridges ++
  new WithDefaultMemModel ++
  new WithFireSimConfigTweaks ++
  new chipyard.LargeBoomAndGemminiRocketConfig

and that boom and gemmini rocket config is core 0 to Boom and 1,2,3,4 to Rocket+Gemmini

Thanks!

PR #776 (2021-06-03): [TracerV] Support Heterogenous Mixes of Cores 
Previously, FIRRTL CheckHighForm would croak because different tracerV target black boxes would have different port definitions. This works around that problem by giving a parameterized name to each black box based on the widths of it's IO. 

Providing widths as parameters in the black box does not suffice to fix this problem, because the _length_ of the Vec of traced instruction can vary from bridge to bridge. 

Also adds some scaladoc. 

Resolves #729.
commit 988d4d7e (2021-06-03): [TracerV] Support Heterogenous Mixes of Cores  (#776)

Resolves #729
closed 2021-06-15 by davidbiancolin