firesim/firesim #752: firesim infrasetup error

zhushaofeng1992, 2021-04-12:
firesim infrasetup
/usr/lib/python2.7/site-packages/paramiko/transport.py:33: CryptographyDeprecationWarning: Python 2 is no longer supported by the Python core team. Support for it is now deprecated in cryptography, and will be removed in the next release.
from cryptography.hazmat.backends import default_backend
FireSim Manager. Docs: http://docs.fires.im
Running: infrasetup

**Building FPGA software driver for FireSim-DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.QuadRocketConfig-F90MHz_BaseF1Config
FPGA software driver build failed. Exiting. See log for details.
You can also re-run 'make DESIGN=FireSim TARGET_CONFIG=DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.QuadRocketConfig PLATFORM_CONFIG=F90MHz_BaseF1Config f1' in the 'firesim/sim' directory to debug this error.**
Fatal error.
Traceback (most recent call last):
File "/home/centos/chipyard/sims/firesim/deploy/firesim", line 334, in
main(args)
File "/home/centos/chipyard/sims/firesim/deploy/firesim", line 282, in main
globals()args.task
File "/home/centos/chipyard/sims/firesim/deploy/firesim", line 95, in infrasetup
runtime_conf.infrasetup()
File "/home/centos/chipyard/sims/firesim/deploy/runtools/runtime_config.py", line 380, in infrasetup
self.firesim_topology_with_passes.infrasetup_passes(use_mock_instances_for_testing)
File "/home/centos/chipyard/sims/firesim/deploy/runtools/firesim_topology_with_passes.py", line 390, in infrasetup_passes
self.pass_build_required_drivers()
File "/home/centos/chipyard/sims/firesim/deploy/runtools/firesim_topology_with_passes.py", line 373, in pass_build_required_drivers
server.get_server_hardware_config().build_fpga_driver()
File "/home/centos/chipyard/sims/firesim/deploy/runtools/runtime_config.py", line 203, in build_fpga_driver
exit(1)
File "/usr/lib64/python2.7/site.py", line 364, in call
raise SystemExit(code)
SystemExit: 1
The full log of this run is:
/home/centos/chipyard/sims/firesim/deploy/logs/2021-04-12--13-14-31-infrasetup-898D5DETR6QW2C4E.log
**********************************
*********************************
i have tried '**make DESIGN=FireSim TARGET_CONFIG=DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.QuadRocketConfig PLATFORM_CONFIG=F90MHz_BaseF1Config f1**' in the 'firesim/sim' directory and got this

cd /home/centos/chipyard/sims/firesim/sim && java -Xmx16G -Dsbt.supershell=false -jar /home/centos/chipyard/sims/firesim/target-design/chipyard/generators/rocket-chip/sbt-launch.jar -Dsbt.sourcemode=true -Dsbt.workspace=/home/centos/chipyard/sims/firesim/target-design/chipyard/tools ";project {file:/home/centos/chipyard/sims/firesim/target-design/chipyard}firechip; runMain chipyard.Generator --target-dir /home/centos/chipyard/sims/firesim/sim/generated-src/f1/FireSim-DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.QuadRocketConfig-F90MHz_BaseF1Config --name firesim.firesim.FireSim.DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.QuadRocketConfig --top-module firesim.firesim.FireSim --legacy-configs firesim.firesim:DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.QuadRocketConfig"
**Error: Unable to access jarfile /home/centos/chipyard/sims/firesim/target-design/chipyard/generators/rocket-chip/sbt-launch.jar**

westre3, 2021-04-12:
I'm experiencing this same issue when attempting to run `firesim infrasetup`. Attached is a copy of my log file.

[2021-04-12--15-47-55-infrasetup-762KXQ9W4TWQB8GH.log](https://github.com/firesim/firesim/files/6298054/2021-04-12--15-47-55-infrasetup-762KXQ9W4TWQB8GH.log)

davidbiancolin, 2021-04-15:
Do you guys have the log from build-setup?

westre3, 2021-04-15:
Here are all of the log files under `firesim/deploy/logs`. Do you need any others?

[firesim_logs.tar.gz](https://github.com/firesim/firesim/files/6321092/firesim_logs.tar.gz)

davidbiancolin, 2021-04-26:
There should be a file in the root of the repository, from when you ran ./build-setup.sh.

davidbiancolin, 2021-05-26:
So it does smell like your initial setup via build-setup.sh or firesim-setup.sh (if using CY as top) did not run properly, and the submodules may not have been ocrrectly initialized. Feel free to reopen if there's still a problem.

closed 2021-05-26 by davidbiancolin