firesim/firesim #771: Check that AFI config matches driver config, and warn on mismatch

jerryz123, 2021-05-27:
Perhaps bake a checksum of the config into the FPGA image, and emit a warning if the checksum reported by the FPGA image doesn't match the config the driver was built with?

timsnyder-siv, 2021-05-27:
Jerry, currently the SHA1 of the toplevel git repo and the `<DESIGN>-<TARGET_CONFIG>-<PLATFORM_CONFIG>` triplet are baked into the AGFI but:
* the driver doesn't get annotated with the same information.  It would be trivial to add as a dynamically generated version string/sha captured in the driver binary as a part of `FireSim-const.h`
* The 'config' of the driver isn't captured in the binary either.  It can be inferred from the directory names where it lives but it would be better to have it also captured in `FireSim-const.h`
* there isn't a runtime check to make sure that the driver was built with the same information
* the software setup for both firesim and chipyard results in '-dirty' changes to the repo (just run `git status` on a chipyard clone after doing setup-firesim.sh), making it impossible to know by the annotated metadata whether it is clean.  It is typically internationalization files that are checked in stale to subrepos and then updated by the build.  This could be cleaned by throwing away those changes in the submodules as a part of the repo setup using very targeted  'git restore' commands.

I have a feeling when you said checksum, you were talking about something other than a git SHA.  A checksum on the serialized "config"?

I've also talked to David about making it so that when you want to run a FireSim AGFI you don't have to build the driver, instead, store the driver in S3 at AGFI build time and then when you want to run, just pull the driver out of S3 using metadata from the AGFI.  However, I think that baking more information into the driver and having runtime checks would still be useful and would continue to apply for non-AWS usage of FireSim if/when that becomes a real thing.

jerryz123, 2021-05-28:
Yes, I meant checksum of the serialized config itself. I agree during software setup and normal usage, both repos end up dirty.
The idea is for a runtime check that the driver matches the FPGA image. When there is a mismatch, the resulting behavior can be strange, and does not immediately suggest a driver-image mismatch.