firesim/firesim #784: Firesim build is  not  working  on  AWS

sadasivan123, 2021-06-12:
Hi,

I  am  trying  to  create BOOM  FPGA using  firesim  build process .  But  I  ran  into  the  following  build  error .  How  to  resolve it.

[localhost] out: Agent pid 3333
[localhost] out: success: firesim.pem available in ssh-agent
[localhost] out: cp -f /home/centos/firesim/sim/generated-src/f1/FireSim-WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.LargeBoomConfig-F65MHz_BaseF1Config/firesim_ila_insert_* /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSim-WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.LargeBoomConfig-F65MHz_BaseF1Config/design/ila_files/
[localhost] out: sed -i "s/\$random/64'b0/g" /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSim-WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.LargeBoomConfig-F65MHz_BaseF1Config/design/ila_files/*
[localhost] out: sed -i "s/\(^ *\)fatal;\( *$\)/\1fatal(0, \"\");\2/g" /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSim-WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.LargeBoomConfig-F65MHz_BaseF1Config/design/ila_files/*
[localhost] out:
[localhost] run: mkdir -p /home/centos/firesim/deploy/results-build/2021-06-12--16-19-02-FireSim-WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.LargeBoomConfig-F65MHz_BaseF1Config-firesim-boom-singlecore-nic-l2-llc4mb-ddr3/
[localhost] out: Agent pid 3333
[localhost] out: success: firesim.pem available in ssh-agent
[localhost] out:
[localhost] run: cp $CL_DIR/design/cl_firesim_generated.sv /home/centos/firesim/deploy/results-build/2021-06-12--16-19-02-FireSim-WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.LargeBoomConfig-F65MHz_BaseF1Config-firesim-boom-singlecore-nic-l2-llc4mb-ddr3/cl_firesim_generated.sv
[localhost] out: Agent pid 3333
[localhost] out: success: firesim.pem available in ssh-agent
[localhost] out:
An error occurred (InvalidParameter) when calling the RunInstances operation: Security group sg-003b2c401755e7f95 and subnet subnet-0bae7172e36f11444 belong to different networks.
This probably means there was no more capacity in this availability zone. Try the next one.
An error occurred (InvalidParameter) when calling the RunInstances operation: Security group sg-003b2c401755e7f95 and subnet subnet-0c3aedf8cfc4a0adf belong to different networks.
This probably means there was no more capacity in this availability zone. Try the next one.
An error occurred (InvalidParameter) when calling the RunInstances operation: Security group sg-003b2c401755e7f95 and subnet subnet-0faee8e52f291ef37 belong to different networks.
This probably means there was no more capacity in this availability zone. Try the next one.
An error occurred (InvalidParameter) when calling the RunInstances operation: Security group sg-003b2c401755e7f95 and subnet subnet-01dbfeac9d6f68a4a belong to different networks.
This probably means there was no more capacity in this availability zone. Try the next one.
An error occurred (InvalidParameter) when calling the RunInstances operation: Security group sg-003b2c401755e7f95 and subnet subnet-0745c4e52d7b4e2c6 belong to different networks.
This probably means there was no more capacity in this availability zone. Try the next one.
An error occurred (InvalidParameter) when calling the RunInstances operation: Security group sg-003b2c401755e7f95 and subnet subnet-0b5c3cf0b64f4a660 belong to different networks.
This probably means there was no more capacity in this availability zone. Try the next one.
we tried all subnets, but there was insufficient capacity to launch your instances
only the following 0 instances were launched
[]
Fatal error.
Traceback (most recent call last):
  File "/home/centos/firesim/deploy/firesim", line 334, in <module>
    main(args)
  File "/home/centos/firesim/deploy/firesim", line 269, in main
    globals()[args.task](buildconfig)
  File "/home/centos/firesim/deploy/firesim", line 140, in buildafi
    globalbuildconf.launch_build_instances()
  File "/home/centos/firesim/deploy/buildtools/buildconfig.py", line 145, in launch_build_instances
    build_farm_prefix)
  File "/home/centos/firesim/deploy/buildtools/buildconfig.py", line 52, in launch_build_instance
    randomsubnet=True)[0]
TypeError: 'NoneType' object has no attribute '__getitem__'
The full log of this run is:
/home/centos/firesim/deploy/logs/2021-06-12--16-19-02-buildafi-FHCCOQBWAHAG0GMB.log
centos@ip-192-168-0-163.ec2.internal:~/firesim$

davidbiancolin, 2021-06-15:
It seems like you may have misconfigured your manager instance. Did you make sure to use the generated firesim security group and subnet?

sadasivan123, 2021-06-16:
Yes,  I have followed  the  firesim documentation .  Is  there any way  I
can  directly  build BOOM V3  from  chipyard ?

Regards,
shaiju

sadasivan123, 2021-06-16:
Here is  the security group  and  VPC

Security groups

   -
   sg-07026b57b1aee7d58 (firesim)
   <https://console.aws.amazon.com/ec2/v2/home?region=us-east-1#SecurityGroup:securityGroupId=sg-07026b57b1aee7d58>

VPC ID
 vpc-09f7ff737a75c795f (firesim)
<https://console.aws.amazon.com/vpc/home?region=us-east-1#vpcs:vpcId=vpc-09f7ff737a75c795f>