firesim/firesim #794: Use fpga-shell as standard shell

sequencer, 2021-06-18:
To make localization more smooth. I think we can considering use fpga-shell to serve as standard shim. This can ease the maintaining burden, and make new platform supports more easy.

So my roadmap is:
1. use my chipyard-like [playground](https://github.com/sequencer/chipsalliance-playground), finish AWS-f1 shell, all of repository are on the master(deb) branches.
2. port to VCU118 for local FPGA support.
3. refactor [cde](https://github.com/sequencer/cde), [diplomacy](https://github.com/sequencer/diplomacy), fpga-shell(tbd) out from SiFive repositories, publish them into maven, use ivy depend on them, I believe this is the most neat way to achieve this job. But I think I need some help from ucb-bar and SiFive to upstream this.

davidbiancolin, 2021-06-18:
I'm very receptive to the idea of making FireSIm more modular and friendly to use in different environments. For instance, some degree of this is required to integrate it more cleanly into our flow here at SiFive. I'm also very receptive to relying on published artifacts instead of building from source. You're proposing pretty enormous changes though... 

I think 1-3 are pretty uncontroversial, though. One question is though, is why reimplement the AWS-shell? And when you say AWS-f1 shell, what do you mean specifically? A diplomatic implementation of everything under firesim_f1.sv in aws-fpga?

sequencer, 2021-06-18:
> why reimplement the AWS-shell?

I think unify different platforms are really necessary for maintenance. Actually we only need MIG and XDMA from F1, so like what we did in VCU118 Shell, we can just implement it to F1 directly. 

> And when you say AWS-f1 shell, what do you mean specifically?  A diplomatic implementation of everything under firesim_f1.sv in aws-fpga?

Yes, maybe a `sifive.fpgashells.shell.xilinx.f1shell.F1Shell` like what we did in `sifive.fpgashells.shell.xilinx.vcu118shell.VCU118Shell`, fully diplomatic.

davidbiancolin, 2021-06-18:
Firstly to avoid  confusion i will refer to the amazon-provided shell as the "SH region" or SH. I will refer to the fpga-shell's notion of a shell as a "diplo-shell". 

I don't think it make much sense to try to implement a diplo-shell for F1 (for FireSim) for a few reasons:

Firstly, the way the AWS flow works, you'll only be able to specify the CL-region. Interface between the SH and CL is locked because they are two different reconfigurable regions. From this perspective, there really is no "XDMA" in the CL, just AXI4 interfaces at the boundary of the CL region. Everything past that boundary, including one of the DRAM controllers, is out of our control. 

Additionally, for the remaining three DRAM interfaces that we do control in the CL, AWS already provides wrappers to instantiate these, i think trying to reimplement them is a waste of effort -- creating a diplomatic wrapper for these wrappers makes more sense. But i don't really know what that gets you. Now, if we wanted to diplomaticize our F1Shim, we should just drive AXI4IOSink nodes whose parameters have been hardcoded to match what the existing AWS template expects.

TL;DR: I think it would just be better if you jumped straight into doing a VCU118 shell.

sequencer, 2021-06-19:
> From this perspective, there really is no "XDMA" in the CL, just AXI4 interfaces at the boundary of the CL region. Everything past that boundary, including one of the DRAM controllers, is out of our control.

I thought on AWS side, they just did some wrapper like sifive/fpga-shells did. My intention was not depending on any aws specific designs, so that the migration flow would be smoother. 

I heard that AWS F1 is alveo U200? I wonder how could aws achieve this locking? For example the PCIe under SLR0 and DDR4 under SLR1 is wrappered by aws shell, I thought in the programming, we are also programming entire fpga? Or AWS magically use the partial programming to achieve this limitation?

davidbiancolin, 2021-06-22:
> I heard that AWS F1 is alveo U200? I wonder how could aws achieve this locking? For example the PCIe under SLR0 and one of DDR4 under SLR1 is wrappered by aws shell, I thought in the programming, we are also programming entire fpga? Or AWS magically use the partial programming to achieve this limitation?

Yeah they use partial reconfiguration (DFX in Vivado parlance). I think we'd want to do the same for Alveo, because otherwise you gotta reboot the system. 

AWS F1 is not exactly a U200, but it is similar.

sequencer, 2021-06-22:
> because otherwise you gotta reboot the system.

Got it. Sounds reasonable, and still curious about why fully reprogramming is required? I thought Linux can reconfigure PCIe devices during the runtime? maybe `modprobe`, `setpci` are something we need to take a look to reload PCIe dynamically?

> AWS F1 is not exactly a U200, but it is similar.

I understand why we need to split Shim and design into two part now. So we need to use fpga-shell as a standard local shell sound reasonable to our use case?
we have different cards for FireSim localization project, including U200 and VCU118, VCU118 already has a fpga-shell implementation, while U200 is similar to F1. What's your suggestion, which card we should adopt firstly?

davidbiancolin, 2021-06-22:
> Got it. Sounds reasonable, and still curious about why fully reprogramming is required? I thought Linux can reconfigure PCIe devices during the runtime? maybe `modprobe`, `setpci` are something we need to take a look to reload PCIe dynamically?

Admittedly, I don't really know the specifics about hot swapping or reconfiguring PCIE devices, but Xilinx provides utilities to do partial reconfiguration over XDMA so i didn't look too much more deeply. I think this forum post hits many of the key details for Alveo:
https://forums.xilinx.com/t5/Alveo-Accelerator-Cards/Anyone-using-TandemPCIe-with-Field-Updates-on-Alveo-Cards/td-p/1076355

> I understand why we need to split Shim and design into two part now. So we need to use fpga-shell as a standard local shell sound reasonable to our use case?
> we have different cards for FireSim [...]

Mmm, i don't really have a strong opinion. Many users have expressed interest in the U200s, but I was going to start building something out for the U250 and porting to the U200 should be easier after the U250 is done. So maybe it makes more sense to do the VCU118? We can talk about it at the FireSim meeting, if you're around. 

For the U250 i was thinking i'd use a Vivado block diagram to do the static shell (since it's static...), and diplomacy to build out the Shim.

davidbiancolin, 2021-06-22:
There appears to be an example design for the VCU118, so maybe that _is_ the way to go.
https://www.xilinx.com/support/documentation/sw_manuals/xilinx2020_2/ug947-vivado-partial-reconfiguration-tutorial.pdf

sequencer, 2021-06-24:
Something interesting.
https://github.com/Xilinx/RapidWright