firesim/firesim #817: Simulating modified kernel and software stack on modified Rocket Core

gdessouky, 2021-07-23:
Can I use FireSim to simulate a modified kernel with kernel module (a modified software stack basically)  on a modified Rocket Core (modified hardware) and collect cycle-accurate measurements for benchmarks like Coremark, etc.?

davidbiancolin, 2021-07-23:
Yup, that's what FireSim was designed to do. :) I'd suggest looking at our FirePerf paper as a case study of what FireSim is capable vis-a-vis full-system performance evaluation. 

We say cycle-exact in our publications because a FireSim simulation should exactly represent the input RTL -- as if you were running in Verilator or VCS (there are some caveats, of course).

gdessouky, 2021-07-23:
"FireSim simulation should exactly represent the input RTL -- as if you were running in Verilator or VCS" - isn't a FireSim simulation the RTL synthesized and prototyped on an FPGA or is it an FPGA-accelerated simulation of the RTL (as if it were running on Verilator but it is accelerated)?

davidbiancolin, 2021-07-23:
Yeah, it's more instructive to think of FireSim as an RTL simulator that happens to use FPGAs to accelerate that simulation. It's not an FPGA prototype, instead it's more closely related to hardware emulators. 

Our papers describe how that is implemented.

davidbiancolin, 2021-12-01:
Please reopen if you have more questions / or post the google groups.

closed 2021-12-01 by davidbiancolin