firesim/firesim #872: launchrunfarm failing to create f1 instance

vargandhi, 2021-11-08:
I’m trying to launch an f1 instance using launchrunfarm, but I keep getting the following error, i.e., lack of capacity in all availability zones. I have no other f1 instances running. Moreover, I was able to successfully create an f1 instance using aws console, so it’s most likely not due to my account running out of its allocation.

I’ve copied the error log below. 

—————————Error Log ———————————————————

2021-11-08 20:08:13,036 [main        ] [INFO ]  FireSim Manager. Docs: http://docs.fires.im
Running: launchrunfarm

2021-11-08 20:08:13,037 [__init__    ] [DEBUG]  {'hwconf_dict': {'firesim-boom-fast-small-quadcore-nic-l2-llc4mb-ddr3': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f70d60863b0>,
                 'firesim-boom-large-quadcore-nic-l2-llc4mb-ddr3': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f70d6086488>,
                 'firesim-boom-medium-singlecore-nic-l2-llc4mb-ddr3': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f70d6086320>,
                 'firesim-boom-quadcore-nic-l2-llc4mb-ddr3': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f70d60863f8>,
                 'firesim-boom-singlecore-large2-nic-l2-llc4mb-ddr3': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f70d6086248>,
                 'firesim-boom-singlecore-nic-l2-llc4mb-ddr3': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f70d6086200>,
                 'firesim-boom-singlecore-no-nic-l2-llc4mb-ddr3': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f70d6086170>,
                 'firesim-boom-small-quadcore-nic-l2-llc4mb-ddr3': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f70d60861b8>,
                 'firesim-rocket-quadcore-nic-l2-llc4mb-ddr3': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f70d6086368>,
                 'firesim-rocket-quadcore-no-nic-l2-llc4mb-ddr3': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f70d6086440>,
                 'firesim-rocket-singlecore-no-nic-l2-llc4mb-ddr3-half-freq-uncore': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f70d60862d8>,
                 'firesim-supernode-rocket-singlecore-nic-l2-lbp': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f70d6086290>}}
2021-11-08 20:08:13,048 [aws_resource] [DEBUG]  i-0cec34b3b2eb6886e
2021-11-08 20:08:13,393 [aws_resource] [DEBUG]  {'Name': 'ime-3'}
2021-11-08 20:08:13,394 [__init__    ] [DEBUG]  {'autocounter_readrate': 0,
 'defaulthwconfig': 'firesim-boom-quadcore-nic-l2-llc4mb-ddr3',
 'disable_asserts': False,
 'f1_16xlarges_requested': 0,
 'f1_2xlarges_requested': 0,
 'f1_4xlarges_requested': 1,
 'linklatency': 6405,
 'm4_16xlarges_requested': 0,
 'netbandwidth': 200,
 'no_net_num_nodes': 2,
 'print_cycle_prefix': True,
 'print_end': '-1',
 'print_start': '0',
 'profileinterval': -1,
 'run_instance_market': 'ondemand',
 'runfarmtag': 'mainrunfarm',
 'spot_interruption_behavior': 'terminate',
 'spot_max_price': 'ondemand',
 'suffixtag': '',
 'switchinglatency': 10,
 'terminateoncompletion': False,
 'topology': 'example_2config',
 'trace_enable': False,
 'trace_end': '-1',
 'trace_output_format': '0',
 'trace_select': '1',
 'trace_start': '0',
 'workload_name': 'nfshield.json',
 'zerooutdram': False}
2021-11-08 20:08:13,401 [get_deploytr] [DEBUG]  Setting deploytriplet by querying the AGFI's description.
2021-11-08 20:08:13,401 [get_afi_for_] [DEBUG]  agfi-00c595c4ae0d05459
2021-11-08 20:08:13,401 [get_afi_for_] [DEBUG]  None
2021-11-08 20:08:17,616 [get_afi_for_] [DEBUG]  {u'FpgaImages': [{u'UpdateTime': datetime.datetime(2021, 10, 1, 4, 21, 42, tzinfo=tzlocal()), u'Name': 'firesim-boom-quadcore-nic-l2-llc4mb-ddr3', u'Tags': [], u'PciId': {u'SubsystemVendorId': '0xfedd', u'VendorId': '0x1d0f', u'DeviceId': '0xf000', u'SubsystemId': '0x1d51'}, u'FpgaImageGlobalId': 'agfi-00c595c4ae0d05459', u'Public': False, u'State': {u'Code': 'available'}, u'ShellVersion': '0x04261818', u'OwnerId': '734394535448', u'FpgaImageId': 'afi-0f77fb25ada9a16dc', u'CreateTime': datetime.datetime(2021, 10, 1, 3, 6, 37, tzinfo=tzlocal()), u'Description': 'firesim-buildtriplet:FireSim-WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.QuadMediumBoomConfig-F65MHz_BaseF1Config,firesim-deploytriplet:FireSim-WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.QuadMediumBoomConfig-F65MHz_BaseF1Config,firesim-commit:72a52523e18443d9ef9a1e2664cbc71a45fc0c57-dirty'}], 'ResponseMetadata': {'RetryAttempts': 0, 'HTTPStatusCode': 200, 'RequestId': '72d34184-4e35-489a-bb9f-29a03c9027ad', 'HTTPHeaders': {'x-amzn-requestid': '72d34184-4e35-489a-bb9f-29a03c9027ad', 'transfer-encoding': 'chunked', 'strict-transport-security': 'max-age=31536000; includeSubDomains', 'vary': 'accept-encoding', 'server': 'AmazonEC2', 'cache-control': 'no-cache, no-store', 'date': 'Mon, 08 Nov 2021 20:08:16 GMT', 'content-type': 'text/xml;charset=UTF-8'}}}
2021-11-08 20:08:17,894 [aws_resource] [DEBUG]  i-0cec34b3b2eb6886e
2021-11-08 20:08:17,955 [aws_resource] [DEBUG]  {'Name': 'ime-3'}
2021-11-08 20:08:21,868 [aws_resource] [DEBUG]  i-0cec34b3b2eb6886e
2021-11-08 20:08:21,934 [aws_resource] [DEBUG]  {'Name': 'ime-3'}
2021-11-08 20:08:31,054 [launch_insta] [INFO ]  An error occurred (InsufficientInstanceCapacity) when calling the RunInstances operation (reached max retries: 4): We currently do not have sufficient f1.4xlarge capacity in the Availability Zone you requested (us-east-1e). Our system will be working on provisioning additional capacity. You can currently get f1.4xlarge capacity by not specifying an Availability Zone in your request or choosing us-east-1a, us-east-1b, us-east-1c, us-east-1d.
2021-11-08 20:08:31,054 [launch_insta] [INFO ]  This probably means there was no more capacity in this availability zone. Try the next one.
2021-11-08 20:08:41,275 [launch_insta] [INFO ]  An error occurred (InsufficientInstanceCapacity) when calling the RunInstances operation (reached max retries: 4): We currently do not have sufficient f1.4xlarge capacity in the Availability Zone you requested (us-east-1c). Our system will be working on provisioning additional capacity. You can currently get f1.4xlarge capacity by not specifying an Availability Zone in your request or choosing us-east-1a, us-east-1b, us-east-1d, us-east-1e.
2021-11-08 20:08:41,276 [launch_insta] [INFO ]  This probably means there was no more capacity in this availability zone. Try the next one.
2021-11-08 20:08:51,745 [launch_insta] [INFO ]  An error occurred (InsufficientInstanceCapacity) when calling the RunInstances operation (reached max retries: 4): We currently do not have sufficient f1.4xlarge capacity in the Availability Zone you requested (us-east-1a). Our system will be working on provisioning additional capacity. You can currently get f1.4xlarge capacity by not specifying an Availability Zone in your request or choosing us-east-1b, us-east-1c, us-east-1d, us-east-1e.
2021-11-08 20:08:51,745 [launch_insta] [INFO ]  This probably means there was no more capacity in this availability zone. Try the next one.
2021-11-08 20:08:52,529 [launch_insta] [INFO ]  An error occurred (Unsupported) when calling the RunInstances operation: Your requested instance type (f1.4xlarge) is not supported in your requested Availability Zone (us-east-1f). Please retry your request by not specifying an Availability Zone or choosing us-east-1a, us-east-1b, us-east-1c, us-east-1d, us-east-1e.
2021-11-08 20:08:52,529 [launch_insta] [INFO ]  This probably means there was no more capacity in this availability zone. Try the next one.
2021-11-08 20:08:59,444 [launch_insta] [INFO ]  An error occurred (InsufficientInstanceCapacity) when calling the RunInstances operation (reached max retries: 4): We currently do not have sufficient f1.4xlarge capacity in the Availability Zone you requested (us-east-1d). Our system will be working on provisioning additional capacity. You can currently get f1.4xlarge capacity by not specifying an Availability Zone in your request or choosing us-east-1a, us-east-1b, us-east-1c, us-east-1e.
2021-11-08 20:08:59,444 [launch_insta] [INFO ]  This probably means there was no more capacity in this availability zone. Try the next one.
2021-11-08 20:09:13,154 [launch_insta] [INFO ]  This probably means there was no more capacity in this availability zone. Try the next one.
[... cut ...]
  File "/home/centos/chipyard/sims/firesim/deploy/firesim", line 166, in launchrunfarm
    runtime_conf.runfarm.launch_run_farm()
  File "/home/centos/chipyard/sims/firesim/deploy/runtools/run_farm.py", line 272, in launch_run_farm
    wait_on_instance_launches(f1_4s, 'f1.4xlarges')
  File "/home/centos/chipyard/sims/firesim/deploy/awstools/awstools.py", line 282, in wait_on_instance_launches
    rootLogger.info("Waiting for instance boots: " + str(len(instances)) + " " + message)
TypeError: object of type 'NoneType' has no len()

davidbiancolin, 2021-11-10:
I'm assuming it doesn't work if you use 2x larges right? 

Generally these problems arise because the manager was misconfigured during initial setup.

vargandhi, 2021-11-10:
Hi David,

I’m getting this error for all f1 instance types, i.e., 2xlarge, 16x large, etc. What do you think could be the possible culprits? Because I have been using my manager instance for months now, and it worked just fine until a few days back. I have two manager instances that were working fine, and now they both are showing the same error. 

I tried creating a new manager instance again, but I am getting the same error. One of the things I observed was that after initializing firesim as a library in chipyard, I run “source sourceme-f1-manager.sh” and get the following output:

-bash: ./env.sh: No such file or directory
success: firesim.pem available in ssh-agent

I’m not sure if a separate env.sh is supposed to be generated during the firesim initialization step that is not getting generated on my instance for some reason, and perhaps I should look into it?

davidbiancolin, 2021-11-10:
Yeah, the env.sh shouldn't be related. That generally means build-setup.sh did not work correctly. 

What version of FireSim are you on? Also, when you launch from the console what region are you launching them in?

vargandhi, 2021-11-11:
I cloned chipyard master branch and initialized firesim submodule as a library as described in the setup instructions, i.e., "./scripts/firesim-setup.sh --fast". Assuming that it clones and initializes a compatible version.  Have the instructions/scripts changed in the recent past? Because in the new manager instance that I have created, I'm unable to build fedora-base.json workload as well and get the following error message:
------ error message --------
2021-11-11 18:45:37,108 [run         ] [DEBUG]  e2fsck: Get a newer version of e2fsck!
2021-11-11 18:45:37,110 [main        ] [ERROR]  Failed to build workload fedora-base.json
-----------------------------

I wonder if this gives you a hint as to what I might be doing wrong because I've followed the setup instructions to the letter. 

My manager instance is in us-east-1 (N.  Virginia) and the error says that all availability zones are out of capacity, i.e.,  launchrunfarm returns a null object. 

Also, how to address the issue of build-setup.sh not working correctly?

davidbiancolin, 2021-11-12:
But you were able to launch f1 instances in the past through the manager right? Is it true that that older clones continue to function correctly?

As for the e2fsck, that might be best posted to Chipyard. Can you build buildroot images?

vargandhi, 2021-11-12:
Hi David,

I had 2 manager instances that were running just fine, but last Saturday onwards I started getting this error. Infact, I created a new manager instance today and launchrunfarm ran successfully once but then I started seeing the same error. Intuitively, it doesn't appear to be a misconfiguration

[image]

davidbiancolin, 2021-11-13:
Huh. I'm kinda miffed. That seems like it might be an AWS-side problem. Can you try a different region with F1 instances?

vargandhi, 2021-11-13:
I had tried creating an f1 instance on us-west-2 earlier this week, but it had failed. Thankfully, it worked just now. However, us-east-1 appears to be still down
[image]

timsnyder-siv, 2022-07-15:
@vargandhi , has this issue gone away for you? If not, feel free to reopen this issue.

closed 2022-07-15 by timsnyder-siv