firesim/firesim #884: firesim infrasetup fails

vargandhi, 2021-11-22:
Hi,

I'm trying to run the fedora-uniform workload on a quad-core rocket config using firesim on ec2, but the infrasetup step fails with the following error:

tldr: -- the error message suggests re-running the hw config, i.e., 'make DESIGN=FireSim TARGET_CONFIG=WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimHighPerfConfigTweaks_chipyard.QuadRocketConfig PLATFORM_CONFIG=WithAutoILA_F90MHz_BaseF1Config f1'

However, when I try to re-run the make command I get the following error: Makefile:4: *** You must source sourceme-f1-manager.sh or env.sh to use this Makefile.  Stop.

I'm a bit confused with this error message because I have source ./env.sh (chipyard) and source sourceme-f1-manager.sh in my bash profile,  and I also sourced them again manually. 

Would really appreciate some help on this, thanks!

-------Infrasetup Error Log--------------
2021-11-22 19:03:08,209 [get_deploytr] [DEBUG]  Setting deploytriplet by querying the AGFI's description.
2021-11-22 19:03:08,209 [get_afi_for_] [DEBUG]  agfi-0935c26b816db7722
2021-11-22 19:03:08,209 [get_afi_for_] [DEBUG]  None
2021-11-22 19:03:10,794 [get_afi_for_] [DEBUG]  {u'FpgaImages': [{u'UpdateTime': datetime.datetime(2021, 6, 11, 12, 41, 55, tzinfo=tzlocal()), u'Name': 'firesim-rocket-quadcore-nic-l2-llc4mb-ddr3', u'Tags': [], u'PciId': {u'SubsystemVendorId': '0xfedd', u'VendorId': '0x1d0f', u'DeviceId': '0xf000', u'SubsystemId': '0x1d51'}, u'FpgaImageGlobalId': 'agfi-0935c26b816db7722', u'Public': True, u'State': {u'Code': 'available'}, u'ShellVersion': '0x04261818', u'OwnerId': '552479941143', u'FpgaImageId': 'afi-05b7ef75aaa22dc42', u'CreateTime': datetime.datetime(2021, 6, 11, 12, 41, 47, tzinfo=tzlocal()), u'Description': 'firesim-buildtriplet:FireSim-WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimHighPerfConfigTweaks_chipyard.QuadRocketConfig-WithAutoILA_F90MHz_BaseF1Config,firesim-deploytriplet:FireSim-WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimHighPerfConfigTweaks_chipyard.QuadRocketConfig-WithAutoILA_F90MHz_BaseF1Config,firesim-commit:b688aff2a5f82fdefd65e7d6d576a2496d02d9b6-dirty'}], 'ResponseMetadata': {'RetryAttempts': 0, 'HTTPStatusCode': 200, 'RequestId': '47640512-e821-45e0-9818-c1579a5050d0', 'HTTPHeaders': {'x-amzn-requestid': '47640512-e821-45e0-9818-c1579a5050d0', 'transfer-encoding': 'chunked', 'strict-transport-security': 'max-age=31536000; includeSubDomains', 'vary': 'accept-encoding', 'server': 'AmazonEC2', 'cache-control': 'no-cache, no-store', 'date': 'Mon, 22 Nov 2021 19:03:10 GMT', 'content-type': 'text/xml;charset=UTF-8'}}}
2021-11-22 19:03:11,251 [bind_real_in] [DEBUG]  Using f1.16xlarge instances with IPs:
[]
2021-11-22 19:03:11,251 [bind_real_in] [DEBUG]  Using f1.4xlarge instances with IPs:
['192.168.1.16']
2021-11-22 19:03:11,251 [bind_real_in] [DEBUG]  Using f1.2xlarge instances with IPs:
[]
2021-11-22 19:03:11,251 [bind_real_in] [DEBUG]  Using m4.16xlarge instances with IPs:
[]
2021-11-22 19:03:11,251 [build_fpga_d] [INFO ]  Building FPGA software driver for FireSim-WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimHighPerfConfigTweaks_chipyard.QuadRocketConfig-WithAutoILA_F90MHz_BaseF1Config
2021-11-22 19:03:11,251 [flush       ] [DEBUG]  [localhost] local: make DESIGN=FireSim TARGET_CONFIG=WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimHighPerfConfigTweaks_chipyard.QuadRocketConfig PLATFORM_CONFIG=WithAutoILA_F90MHz_BaseF1Config f1
2021-11-22 19:03:11,258 [flush       ] [DEBUG]  Warning: local() encountered an error (return code 1) while executing 'make DESIGN=FireSim TARGET_CONFIG=WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimHighPerfConfigTweaks_chipyard.QuadRocketConfig PLATFORM_CONFIG=WithAutoILA_F90MHz_BaseF1Config f1'
2021-11-22 19:03:11,258 [build_fpga_d] [DEBUG]  [localhost]
2021-11-22 19:03:11,258 [build_fpga_d] [DEBUG]  [localhost] ./sourceme-f1-manager.sh: line 11: ./env.sh: No such file or directory
2021-11-22 19:03:11,258 [build_fpga_d] [INFO ]  FPGA software driver build failed. Exiting. See log for details.
2021-11-22 19:03:11,258 [build_fpga_d] [INFO ]  You can also re-run 'make DESIGN=FireSim TARGET_CONFIG=WithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimHighPerfConfigTweaks_chipyard.QuadRocketConfig PLATFORM_CONFIG=WithAutoILA_F90MHz_BaseF1Config f1' in the 'firesim/sim' directory to debug this error.
2021-11-22 19:03:11,258 [<module>    ] [ERROR]  Fatal error.
Traceback (most recent call last):
  File "/home/centos/chipyard/sims/firesim/deploy/firesim", line 356, in <module>
    main(args)
  File "/home/centos/chipyard/sims/firesim/deploy/firesim", line 304, in main
    globals()[args.task](simconf)
  File "/home/centos/chipyard/sims/firesim/deploy/firesim", line 95, in infrasetup
    runtime_conf.infrasetup()
  File "/home/centos/chipyard/sims/firesim/deploy/runtools/runtime_config.py", line 380, in infrasetup
    self.firesim_topology_with_passes.infrasetup_passes(use_mock_instances_for_testing)
  File "/home/centos/chipyard/sims/firesim/deploy/runtools/firesim_topology_with_passes.py", line 390, in infrasetup_passes
    self.pass_build_required_drivers()
  File "/home/centos/chipyard/sims/firesim/deploy/runtools/firesim_topology_with_passes.py", line 373, in pass_build_required_drivers
    server.get_server_hardware_config().build_fpga_driver()
  File "/home/centos/chipyard/sims/firesim/deploy/runtools/runtime_config.py", line 203, in build_fpga_driver
    exit(1)
  File "/usr/lib64/python2.7/site.py", line 364, in __call__
    raise SystemExit(code)
SystemExit: 1
2021-11-22 19:03:11,259 [<module>    ] [INFO ]  The full log of this run is:
/home/centos/chipyard/sims/firesim/deploy/logs/2021-11-22--19-03-07-infrasetup-WOWBHH5RAZXNVHD0.log

davidbiancolin, 2021-11-22:
This looks like build-setup.sh didn't run correctly, and so env.sh wasn't generated. If that's the case, there should be a build-setup log at the top of the repo you should read through to try to figure out what happened.

davidbiancolin, 2021-12-01:
@vargandhi I'm going to close this. Reopen if you find and issue with build setup you can't resolve.

closed 2021-12-01 by davidbiancolin