firesim/firesim #894: Bucket name forbidden

mbelda, 2021-12-02:
Hello, 

I am currently using the last version of this repo and I keep getting this error when I try to run "firesim buildafi".

You tried to access a bucket that is Forbidden. This probably means that someone else has taken the name already.
The full exception is printed below:
____________________________________________________________
ClientError(u'An error occurred (403) when calling the HeadBucket operation: Forbidden',)
Fatal error.
Traceback (most recent call last):
  File "/home/centos/firesim/deploy/firesim", line 356, in <module>
    main(args)
  File "/home/centos/firesim/deploy/firesim", line 291, in main
    globals()[args.task](buildconfig)
  File "/home/centos/firesim/deploy/firesim", line 120, in buildafi
    auto_create_bucket(globalbuildconf.s3_bucketname)
  File "/home/centos/firesim/deploy/awstools/awstools.py", line 318, in auto_create_bucket
    assert False
AssertionError

I have ensured that my bucket identifier is unique trying different identifiers with my name and random numbers involved.

davidbiancolin, 2021-12-02:
Can you confirm that you can push and get objects from the bucket manually using the AWS CLI?

mbelda, 2021-12-02:
I'm afraid I don't know how to push and get manually with CLI. Can you post the steps to follow?

I got an old repo and it worked properly but with this instance and repo version I have not used buildafi until now. 

I have modified the config_build_recipes.ini and config_build.ini files and added my custom configs, but I think they are OK because I have created them previously on my other instance.

davidbiancolin, 2021-12-02:
https://docs.aws.amazon.com/cli/latest/reference/s3/cp.html There are some examples at the bottom.

When you say latest, you mean master right?

If this is new instance it is ossible aws configure didn't run correctly (when you ran firesim managerinit).

mbelda, 2021-12-02:
I checked that my firesim managerinit didn't run properly.

How can I rerun the command so that I change the Access ID, key ID, region and format file?
When I rerun it it just doesn't show me the option to change the values I entered.

mbelda, 2021-12-02:
I run "aws configure" and I changed the parameter. Now everything is working fine.
Thank you very much. 
I close the issue.

Regards, MªJosé.

closed 2021-12-02 by mbelda