firesim/firesim #897: Error building AGFI for BOOM multicore

mbelda, 2021-12-13:
Hello,
I am having issues building an AGFI with the newer version of the repo for a BOOM 4-core system.
It throws an error with little information.

My config is the following:
[firesim-boom-quadcore-no-nic-l2-llc4mb-ddr3-MCRams-50MHz]
DESIGN=FireSim
TARGET_CONFIG=DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimTestChipConfigTweaks_chipyard.QuadLargeBoomConfig
PLATFORM_CONFIG=MCRams_F50MHz_BaseF1Config
instancetype=z1d.2xlarge
deploytriplet=None

I have also tried adding WithAutoILA. And I have successfully generated other architectures with 4-Rocket multicore, so i think is not a problem of the environment.

Also, I followed this issue https://github.com/firesim/firesim/issues/484 but it it keeps throwing the same error.

And my log file for the run can be find here https://drive.google.com/file/d/1MI_7e-_9ypgVsa6cAVCK0MB2pTEjDy-x/view?usp=sharing

davidbiancolin, 2021-12-14:
I can't open that link, can please attach the tail of the log for that build? 

I suspect you're running out of resources. You should look at the instance-multithreading optimization as it's the only way you'll probably be able to fit. https://docs.fires.im/en/latest/Golden-Gate/Resource-Optimizations.html

mbelda, 2021-12-14:
Here it is the log. I am going to see that part of the documentation. 

[2021-12-10--10-18-09-buildafi-97Y0OCLHZESETX2Y(1).log](https://github.com/firesim/firesim/files/7710431/2021-12-10--10-18-09-buildafi-97Y0OCLHZESETX2Y.1.log)

Thanks.

mbelda, 2021-12-14:
I was using the optimization MCRams, but not the second one.
Nevertheless I just tried to build the architecture with also the MTModels optimization and it stills gives a Java.lang.OutOfMemoryError. Here you have the new log of this run.
[2021-12-14--11-40-35-buildafi-Y5AU13WIM9D89X3R.log](https://github.com/firesim/firesim/files/7711472/2021-12-14--11-40-35-buildafi-Y5AU13WIM9D89X3R.log)