firesim/firesim #901: Metasimulation broken under Modern VCS Releases

davidbiancolin, 2021-12-15:
It appears that a some version between after L-2016.06-SP2-4_Full64 and M-2017.03-SP2_Full64, VCS started clobbering older allocations to the heap made by the driver (we enter through the driver before switching into vcs_main). This manifests first in segmentation faults when referencing mmio_t and mm_t datastructures used in the DPI call to switch back to the driver context.

I've reproduced this on millennium (Ubuntu) and sifive (RHEL) machines for a variety of different VCS versions.  I'm going to try reworking VCS-based metasimulation setup to use VCS's main as the entrypoint for the program instead of the driver, but i'm worried that later heap allocations in either context could create problems. 

h/t @aswaterman

timsnyder-siv, 2022-08-06:
@davidbiancolin you said that you're no longer seeing this, right?

davidbiancolin, 2022-08-06:
i have vcs metasimulators running with some change to how the mm_t and mmio_t pointers are handled, but i'm not confident that this problem is resolved yet.