firesim/firesim #902: Having trouble understanding how AutoCounter works

haeunlee22, 2021-12-18:
Hello,

I am having some trouble understanding how AutoCounter works. I am trying to measure how many floating point operations and LLC misses there were while running my workload. As mentioned in the firesim documentation, I have built a new image as follows so that I could use an AutoCounter. 

```
[firesim-hwacha-rocket-quadcore-no-nic-ddr3-no-autoila-autocounter]
DESIGN=FireSim
TARGET_CONFIG=DDR3FRFCFS_WithDefaultFireSimBridges_WithFireSimTestChipConfigTweaks_chipyard.HwachaQuadRocketConfig
PLATFORM_CONFIG=F90MHz_BaseF1Config_WithAutoCounter
instancetype=z1d.2xlarge
deploytriplet=None
```

Then I have launched runfarm and executed my workload. I was expecting to see sim_slot_<slot #> directory anywhere under my home directory on the F1 instance. Also, when I terminated my F1 instance, I could only see uartlog under ~/firesim/deploy/result-workloads. 

I thought AutoCounter by default would measure some metrics such as data cache misses. Here are my questions. 

1. Do I have to manually add all metrics that I want to measure by myself? If yes where are the list of metrics that I can ask the counter to measure? If "midas.targetutils.PerfCounter(en_clock, "gate_clock", "Core clock gated")" given in documentation does this job, where should this line be placed?

2. I am running binary file cross-compiled with gcc. At the moment, I don't need tracing or function level performance analyzes. Do I need to add anything to the source file I am trying to run?

I am sorry if I am asking trivial questions. I tried to find solutions myself, but couldn't really find a way out.

Thank you!

davidbiancolin, 2021-12-18:
Hello! 

So firstly, i'd suggest prefixing your config string with new options, so: "WithAutoCounter_F90MHz_BaseF1Config". In this case it should work anyways. Second, you'll probably need to add the autocounter output file to your workload.ini so that it gets copied back. See: https://docs.fires.im/en/latest/Advanced-Usage/Workloads/Defining-Custom-Workloads.html#uniform-workload-json

> 1. Do I have to manually add all metrics that I want to measure by myself? If yes where are the list of metrics that I can ask the counter to measure? If "midas.targetutils.PerfCounter(en_clock, [...]

Yes, currently you do. It's easiest to add those annotations in the Chisel sources that define events you care about. You can open an topic in the chipyard google groups if you want to know how to annotate specific events. 

> 2. I am running binary file cross-compiled with gcc. At the moment, I don't need tracing or function level performance analyzes. Do I need to add anything to the source file I am trying to run?

You don't have to chnage anything about your application to use autocounter. The results are collected by the simulator automatically, without interfering with your program execution.

haeunlee22, 2021-12-26:
Thank you for your comment and sorry for late thanks.
I am new to Chisel  so was not able to exactly grasp your suggestions.
I will open a topic in google groups soon to learn how to use auto counter!

haeunlee22, 2021-12-28:
Thank you for your answer and sorry for reopening the issue :(

I would like to ask with example whether it is correct to use autocounter like below.
I am very new to chisel, so please bear with me if I am making stupid mistakes.
Assume I have a source code like below at firesim, having counter that increments when function **increment** is called. 
I have defined an abstract class to write simple code. 

```
abstract class Example() extends Module {
    val counter = RegInit(0.U(8.W))
    def increment() : Unit = {
        counter := counter + 1.U
    }
}
```

Now, if I want to insert autocounter, would adding line like below give me output file after I power off f1 instance?
I have edited workload json file like below as suggested.

**workload json**
```
{
  "name" : "hwacha_fedora",
  "base" : "fedora-base.json",
  "overlay" : "overlay",
  "common_simulation_outputs" : ["uartlog", "memory_stats*.csv", "AUTOCOUNTER*"],
  "jobs" : [
    {
      "name" : "VA"
    },
  ]
}
```

**source code**
```
import midas.targetutils.PerfCounter

abstract class Example() extends Module {
    val counter = RegInit(0.U(8.W))
    PerfCounter(counter, "counter_var", "Counts number of increment")
    def increment() : Unit = {
        counter := counter + 1.U
    }
}
```

Also, I think I can set second and third parameter of PerCounter with arbitrarily string while the first parameter should be actual value to be tracked. But I am not too sure whether I am understanding correctly. 

Thank you!

closed 2021-12-29 by haeunlee22