firesim/firesim #905: Can FireSim be deployed to an on-premise cluster?

howardlau1999, 2022-01-05:
We have serveral FPGA servers running and would like to use FireSim on these servers instead of AWS.

SihaoLiu, 2022-01-06:
I am also expecting this feature. I believe this is related to this PR https://github.com/firesim/firesim/pull/848, but for some reason it is closed

davidbiancolin, 2022-01-12:
We are actively working on this. I think we'll have something running on Alveo cards for the next release.

hyviquel, 2022-05-25:
@davidbiancolin any news on this? what is the current status of Alveo cards outside of AWS?

davidbiancolin, 2022-05-25:
Yes, we do plan to do that. What sorts of model are you thinking of? 

We've been trying to re-engineer the interface between the host and the rest of the simulator to make that process cleaner, i'm not sure if all of those changes will be complete in time for the release.

davidbiancolin, 2022-05-25:
For FireSim to work reasonably well, you're going to want a low-ish latency link between an x86 host and your FPGA. The existing examples will use PCI-E. What boards are you using? 

In the past we supported MPSoCs, i think it should be possible to support that easily in the future, but we haven't quite figured out the cross-compilation story for non x86 deployment hosts.

howardlau1999, 2022-05-25:
> For FireSim to work reasonably well, you're going to want a low-ish latency link between an x86 host and your FPGA. The existing examples will use PCI-E. What boards are you using?

We have serveral Zynq boards.

> In the past we supported MPSoCs, i think it should be possible to support that easily in the future, but we haven't quite figured out the cross-compilation story for non x86 deployment hosts.

I think our boards are MPSoCs. I think since you supported MPSoCs in the past, someone in the community may find out how to cross-compile for non-x86 hosts.