firesim/firesim #1000: Buildafi fails with medium boom

mbelda, 2022-03-29:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

Release: 1.13.0
Hash: a6a6a6

### OS Setup

LSB Version:	:core-4.1-amd64:core-4.1-noarch
Distributor ID:	CentOS
Description:	CentOS Linux release 7.9.2009 (Core)
Release:	7.9.2009
Codename:	Core

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

I tried to run buildafi to generate a design with one medium boom core and a hwacha vector procesor specified with this config:
class HwachaMediumBoomConfig extends Config(
  new chipyard.config.WithHwachaTest ++
  new hwacha.DefaultHwachaConfig ++                              // use Hwacha vector accelerator
  new boom.common.WithNMediumBooms(1) ++

Also tried to generate a more complex design with one medium boom, one rocket, Hwacha and Gemmini and it gives the same error. Nevertheless, a config with just two medium booms is generated fine.

The error it gave is very uncleared. I attach the log of the run.

[2022-03-29--19-14-27-buildafi-IN9B5HVSNBPK28IT.log](https://github.com/firesim/firesim/files/8374564/2022-03-29--19-14-27-buildafi-IN9B5HVSNBPK28IT.log)

### Expected Behavior

I would like to get the design generated correctly.

davidbiancolin, 2022-03-29:
Can you put the relevant bit of the log in the issue.

mbelda, 2022-03-30:
Here is where the log start showing the errors.

2022-03-29 19:15:11,953 [flush       ] [INFO ]  [localhost] out: [info] running chipyard.Generator --target-dir /home/centos/firesim-copy/sim/generated-src/f1/FireSim-DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimTestChipConfigTweaks_chipyard.HwachaMediumBoomConfig-MCRams_F60MHz_BaseF1Config --name firesim.firesim.FireSim.DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimTestChipConfigTweaks_chipyard.HwachaMediumBoomConfig --top-module firesim.firesim.FireSim --legacy-configs firesim.firesim:DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimTestChipConfigTweaks_chipyard.HwachaMediumBoomConfig
2022-03-29 19:15:18,715 [flush       ] [INFO ]  [localhost] out: L2 InclusiveCache Client Map:
2022-03-29 19:15:18,716 [flush       ] [INFO ]  [localhost] out: 	0 <= blkdev-tracker0
2022-03-29 19:15:18,716 [flush       ] [INFO ]  [localhost] out: 	1 <= serial-tl
2022-03-29 19:15:18,716 [flush       ] [INFO ]  [localhost] out: 	2 <= Core 0 DCache
2022-03-29 19:15:18,716 [flush       ] [INFO ]  [localhost] out: 	3 <= Core 0 ICache
2022-03-29 19:15:18,716 [flush       ] [INFO ]  [localhost] out: 	4 <= Core 0 ICache
2022-03-29 19:15:18,716 [flush       ] [INFO ]  [localhost] out: 	5 <= HwachaSMU
2022-03-29 19:15:18,716 [flush       ] [INFO ]  [localhost] out: 	6 <= HwachaVMU
2022-03-29 19:15:18,718 [flush       ] [INFO ]  [localhost] out: 
2022-03-29 19:15:21,453 [flush       ] [INFO ]  [localhost] out: [error] (run-main-0) java.lang.reflect.InvocationTargetException
2022-03-29 19:15:21,471 [flush       ] [INFO ]  [localhost] out: [error] java.lang.reflect.InvocationTargetException
2022-03-29 19:15:21,472 [flush       ] [INFO ]  [localhost] out: [error] 	at ... ()
2022-03-29 19:15:21,472 [flush       ] [INFO ]  [localhost] out: [error] 	at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:36)
2022-03-29 19:15:21,472 [flush       ] [INFO ]  [localhost] out: [error] 	at ... ()
2022-03-29 19:15:21,472 [flush       ] [INFO ]  [localhost] out: [error] 	at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
2022-03-29 19:15:21,472 [flush       ] [INFO ]  [localhost] out: [error] Caused by: java.lang.IllegalArgumentException: requirement failed
2022-03-29 19:15:21,472 [flush       ] [INFO ]  [localhost] out: [error] 	at scala.Predef$.require(Predef.scala:268)
2022-03-29 19:15:21,472 [flush       ] [INFO ]  [localhost] out: [error] 	at boom.ifu.ICacheModule.<init>(icache.scala:144)
2022-03-29 19:15:21,472 [flush       ] [INFO ]  [localhost] out: [error] 	at boom.ifu.ICache.$anonfun$module$1(icache.scala:47)
2022-03-29 19:15:21,472 [flush       ] [INFO ]  [localhost] out: [error] 	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
2022-03-29 19:15:21,473 [flush       ] [INFO ]  [localhost] out: [error] 	at boom.ifu.ICache.module$lzycompute(icache.scala:47)
2022-03-29 19:15:21,473 [flush       ] [INFO ]  [localhost] out: [error] 	at boom.ifu.ICache.module(icache.scala:47)
2022-03-29 19:15:21,473 [flush       ] [INFO ]  [localhost] out: [error] 	at boom.ifu.ICache.module(icache.scala:42)
2022-03-29 19:15:21,473 [flush       ] [INFO ]  [localhost] out: [error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$3(LazyModule.scala:278)
2022-03-29 19:15:21,473 [flush       ] [INFO ]  [localhost] out: [error] 	at chisel3.Module$.do_apply(Module.scala:53)
2022-03-29 19:15:21,477 [flush       ] [INFO ]  [localhost] out: [error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$2(LazyModule.scala:278)
2022-03-29 19:15:21,477 [flush       ] [INFO ]  [localhost] out: [error] 	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
2022-03-29 19:15:21,477 [flush       ] [INFO ]  [localhost] out: [error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$1(LazyModule.scala:278)
2022-03-29 19:15:21,478 [flush       ] [INFO ]  [localhost] out: [error] 	at scala.collection.immutable.List.flatMap(List.scala:338)
2022-03-29 19:15:21,478 [flush       ] [INFO ]  [localhost] out: [error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate(LazyModule.scala:276)
2022-03-29 19:15:21,478 [flush       ] [INFO ]  [localhost] out: [error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate$(LazyModule.scala:273)
2022-03-29 19:15:21,478 [flush       ] [INFO ]  [localhost] out: [error] 	at freechips.rocketchip.diplomacy.LazyModuleImp.instantiate(LazyModule.scala:335)
2022-03-29 19:15:21,478 [flush       ] [INFO ]  [localhost] out: [error] 	at freechips.rocketchip.diplomacy.LazyModuleImp.$anonfun$x$21$1(LazyModule.scala:337)
2022-03-29 19:15:21,478 [flush       ] [INFO ]  [localhost] out: [error] 	at chisel3.internal.plugin.package$.autoNameRecursivelyProduct(package.scala:48)
2022-03-29 19:15:21,481 [flush       ] [INFO ]  [localhost] out: [error] 	at freechips.rocketchip.diplomacy.LazyModuleImp.<init>(LazyModule.scala:337)
2022-03-29 19:15:21,481 [flush       ] [INFO ]  [localhost] out: [error] 	at boom.ifu.BoomFrontendModule.<init>(frontend.scala:324)
2022-03-29 19:15:21,482 [flush       ] [INFO ]  [localhost] out: [error] 	at boom.ifu.BoomFrontend.$anonfun$module$1(frontend.scala:299)
2022-03-29 19:15:21,482 [flush       ] [INFO ]  [localhost] out: [error] 	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
[... cut ...]
2022-03-29 19:15:22,098 [flush       ] [INFO ]  [localhost] out: make: *** [/home/centos/firesim-copy/sim/generated-src/f1/FireSim-DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimTestChipConfigTweaks_chipyard.HwachaMediumBoomConfig-MCRams_F60MHz_BaseF1Config/firesim.firesim.FireSim.DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimTestChipConfigTweaks_chipyard.HwachaMediumBoomConfig.fir] Error 1
2022-03-29 19:15:22,101 [flush       ] [INFO ]  [localhost] out: 
2022-03-29 19:15:22,101 [flush       ] [INFO ]  Fatal error: run() received nonzero return code 2 while executing!

davidbiancolin, 2022-03-30:
Hmm, this is probably a question better posed the Chipyard google group since this a problem with RTL elaboration. But @jerryz123 do you have any insight?

jerryz123, 2022-03-30:
I believe Hwacha is only compatible with LargeBoom currently. (That is a bug that deserves to be fixed)