firesim/firesim #1015: Use SBT thin-client for CI

abejgonzalez, 2022-04-11:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### Feature Description

After the Conda work is done/merged-in it would be good to experiment with the CI and see how much time is saved using the SBT thin client (which should already be enabled w/ FireSim). The only main change that needs to be done is that the Conda install script needs to install the SBT script (if you just use the thin client with the `sbt-launch.jar` not wrapped in the proper SBT script, it will exit 0 no matter what).

### Motivating Example

N/A

timsnyder, 2022-04-11:
You should be able to just add `sbt` into the list of conda packages in `machine-launch-script.sh` and it will install https://anaconda.org/conda-forge/sbt/files.  Let me know if you encounter trouble with it and I'll try to help out.

abejgonzalez, 2022-04-12:
The plan is to do this once #986 is in `main`.

closed 2024-08-07 by abejgonzalez