firesim/firesim #1039: Autocomplete Broken - Git / FireSim Manager

abejgonzalez, 2022-04-22:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

Release: N/A
Hash: 68b0938

### OS Setup

Whatever AMI is specified on `main`. Default setup.

### Other Setup

Default setup of a FireSim manager.

### Current Behavior

There is no autocomplete for things like `git` and the `firesim` command. 

### Expected Behavior

I expect to have autocomplete for things like `git` and the `firesim` command.

### Other Information

I think with the bump to conda (and change in `machine-launch-script.sh`) this broke - #986

timsnyder-siv, 2022-04-23:
Can you share the output you get from `complete -p` in a shell where completion was working the way you wanted it to and what you see now?

timsnyder-siv, 2022-04-23:
Looking into this a bit more, the old yum-based machine-launch-script.sh installed `bash-completion` from yum and this adds the toplevel `/etc/profile.d/bash_completion.sh` which enables all of the completions in `/etc/bash_completion.d/`.

There is a `bash-completion` conda package that can be added into our environment and that will make git start working.  I'm not sure what it looks like for firesim completion to be "working" because I've never had it work for me.  I'll open a PR with some changes that you can try.

PR #1041 (2022-04-28): add bash-completion and install argcomplete global into it

#### Related PRs / Issues

fixes #1039 

#### UI / API Impact

Should make bash-completion work in the conda environment.

#### Verilog / AGFI Compatibility

### Contributor Checklist
- [X] Is this PR's title suitable for inclusion in the changelog and have you added a `changelog:<topic>` label?
- [x] Did you add Scaladoc/docstring/doxygen to every public function/method?
- [x] Did yo
closed 2022-04-28 by timsnyder-siv