firesim/firesim #1051: Adding "firesim kill notion" in the doc

xingyuli9961, 2022-05-03:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### Feature Description

Currently, after a" firesim runworkload" is launched, a user can ctrl+c to exit the simulation. However, this doesn't really stops the simulation. I tried to call a new "infrasetup" but the host is still occupied. Another "firesim kill' is needed to actually kill the simulation, so the user may re-use the same run farm in future simulations. 
The commands already exists, but not mentioned in the documentation of Firesim at an obvious place. I would like this "firesim kill" command to be mentioned in the https://docs.fires.im/en/stable/Running-Simulations-Tutorial/Running-a-Cluster-Simulation.html#setting-up-the-manager-configuration, so it would be easier for users to know how to actually stop a simulation and reuse the run farm.

### Motivating Example

This should have been already discussed in the description above.

timsnyder-siv, 2022-07-15:
@xingyuli9961 thanks for pointing this out.  We should try to document this soon.

timsnyder-siv, 2022-07-18:
It might be good for the SIGINT handler of `firesim runworkload` to mention that any running simulations have not been killed and that the clusters are still operating, only the monitoring of the workload has been killed.  This message should probably also be printed for any other abnormal termination of the monitoring loop.