firesim/firesim #1068: Where-To-Run Leftover Feature Tracker

abejgonzalez, 2022-06-04:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### Feature Description

- Look into the naming of AWS EC2 across the code base (i.e. should it be called f1, AWSEC2, aws_ec2, etc): https://github.com/firesim/firesim/pull/1028#discussion_r887451085
- Look at padding the hostname/ip in the simulation status printout: https://github.com/firesim/firesim/pull/1028#discussion_r887364946
- Test networked sims w/ Externally Provisioned run farm
- In the documentation, mention how the overrides work (including you can't override the type of runfarm). https://github.com/firesim/firesim/pull/1028#discussion_r885857553

### Motivating Example

N/A

abejgonzalez, 2022-06-07:
@ffard-lbl

abejgonzalez, 2022-06-08:
For "Look into the naming of AWS EC2 across the code base (i.e. should it be called f1, AWSEC2, aws_ec2, etc)":

Right now there are a couple things that need consistent names:

Build Farms: AWSEC2F1, ExternallyProvisioned
Run Farms: AWSEC2, ExternallyProvisioned
BitBuilders: F1, Vitis
DeployManagers: EC2DeployManager, VitisDeployManager
(in addition the yaml files need to match)

For the AWS case, what I've been adhering to is if the item (build/run farm,bitbuilder) needs FPGAs then it has F1 appended to it (thus AWSEC2F1). Since the run farm isn't strictly FPGA instances then it is just AWSEC2.

How does this sound for everything? This generalizes the name of anything related to AWS to "AWS" and makes the YAML file match the same case.

Uses anything on AWS
- Classname = `AWS`
- Yamlname = `AWS.yaml`

ExternallyProvisioned
- Classname = `ExternallyProvisoned`
- Yamlname = `ExternallyProvisioned.yaml`

Bit Builders:
- Classname = `AWS`, `Vitis
- Yamlname = `AWS.yaml`, `Vitis.yaml`

Deploy Managers:
- Classname = `AWS`, `Vitis`

davidbiancolin, 2022-06-08:
Oh i think we should avoid the use of the term `Manager` in any manager-related literature / structures / code. This will avoid the confusing AF situation we have with the driver.

timsnyder-siv, 2022-07-03:
@abejgonzalez, looking at some of the new classes (e.g. `BitBuilder`) seem to be missing `__str__` and `__repr__` implementations.  I know we talked about including them for every class in one of the early PR's.  It might be good to add this as a cleanup item to your checkboxes in the OP.

timsnyder-siv, 2022-07-05:
# Add URI's for xclbins/other bitstreams

I'm relearning pandas and came across the libraries that I was thinking we should use to implement this (because they are used by pandas and a bunch of other libraries and thus should hopefully be battle hardened).  I'd use https://filesystem-spec.readthedocs.io/en/latest/usage.html#higher-level.  

Don't get distracted by the lower-level interface that they start out describing at the top of the page.

Adding use of this library will enable users to store bitstreams in many ways in addition to local files without us having to do any work.

We could start out only using this for the bitstreams and it could be extended to the rest of the fileio of the manager fairly easily if we think that's useful.  For example, you could store the hwdb itself in a S3 bucket and be able to access it from your AWS manager and local FPGA machines. It would be really dope to see if there are C/C++ bindings and/or create some and make it so the driver code could pull things from URLs in a distributed runfarm instead of the manager having to push the artifacts from a single worker.  Would make it easier to scale up FireSim simulations to 1000's of workers because one could leverage the cloud provider's distributed infrastructure.

I'll still try to pull together a PR today or tomorrow for this with the bitstreams in the hwdb.yaml.

abejgonzalez, 2022-07-13:
Wow they support alot of backend filesystems. I look forward to hosting our config files on Google drive! 
https://github.com/fsspec/filesystem_spec/blob/a8cfd9c52a20c930c67ff296b60dbcda89d64db9/fsspec/registry.py#L87-L205

timsnyder-siv, 2022-07-13:
Took me longer to get around to looking at `fsspec` but I started this afternoon.  The docs aren't super great, especially since we want to use the filesystem abstraction to copy stuff from any of the supported url protocols to a local file, that isn't documented much at all.  However, I was able to figure out how to copy files from remote to local with:
```
from fsspec.core import url_to_fs
from paramiko import RSAKey
first_url = 'ssh://192.168.0.133/home/centos/awsver.txt'
pkey = RSAKey.from_private_key_file('/home/centos/firesim.pem')
fs, rpath = url_to_fs(first_url, pkey=pkey)
fs.get(rpath, '/tmp/awsver.txt')
```

I think I only needed to create and pass the `RSAKey` argument because I was messing around in a Jupyter notebook that didn't have the key in a SSHAgent.  I'll try it again in an actual script without explicitly adding the key and see if I can get a PR submitted today, including updates to the docs.