firesim/firesim #1080: Bump paramiko-ng to get rid of blowfish deprecation warning

sagark, 2022-06-13:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

Release: 1.14.0

### OS Setup

ec2 f1

### Current Behavior

This warning prints at the start of every manager invocation: https://github.com/paramiko/paramiko/issues/2038

However, fab-classic uses paramiko-ng, which does not have the fix listed above yet.

We temporarily implemented #1079, which hides this warning until 2022-08-31.

### Expected Behavior

No warning printing.

timsnyder, 2022-06-13:
Filed https://github.com/ploxiln/paramiko-ng/issues/135 to ask about this.

timsnyder, 2022-07-05:
Fixed in https://github.com/ploxiln/paramiko-ng/pull/136.  I asked about getting it released in that PR.

timsnyder-siv, 2022-07-15:
Still no update about getting the merged fix into a release.