firesim/firesim #1112: Get rid of old failing "use-fpgas" and "build-afis" environments that appear on main repo page

sagark, 2022-06-27:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

Release: 1.13.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` and `lsb_release -a`

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

a

### Expected Behavior

a

davidbiancolin, 2022-07-05:
@abejgonzalez I could basically just go ahead and remove these right?

abejgonzalez, 2022-07-13:
Done

closed 2022-07-13 by abejgonzalez