firesim/firesim #1127: Implementation of firesim in zynq zedboard

qiuhaojie, 2022-07-11:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### Feature Description

We want to continue implementing firesim in the zynq zedboard, so we want to understand where the current state of the zynq branch is stalled?

### Motivating Example

We tested the zynq branch in Zedboard and found that PS accessing PL via AXI bus would system hang, suspecting that there is a compatibility problem between rocket-chip-master\src\main\scala\amba\axi4\Xbar.scala and ARM's AXI bus timings.

davidbiancolin, 2022-07-27:
We have no foreseeable plans to support zynq because of the complexity of managing cross-builds for ARM-based platforms. This might change in the future.

qiuhaojie, 2022-07-29:
yes,but now there is a problem when we start Linux. Is there a DMA in the SOC system when we start it; At present, the fpgatop produced by zynq seems to have a DMA interface, but there is no DMA controller?We started Linux and got stuck in this position. Is there a problem with the file system or with the peripherals?
r-base-bin1374665:/xsim# ./FireSim-zc706 +permissive +mm_relaxFunctionalModel_0=0 +mm_writeMaxReqs_0=10 +mm_readMaxReqs_0=10 +mm_writeLatency_0=30 +mm_readLatency_0=30 +blkdev0=br-base.img +permissive-off bUART0 is here (stdin/stdout).
command line for program 0. argc=9:
+permissive +mm_relaxFunctionalModel_0=0 +mm_writeMaxReqs_0=10 +mm_readMaxReqs_0=10 +mm_writeLatency_0=30 +mm_readLatency_0=30 +blkdev0=br-base.img +permissive-off br-base-bin
TraceRV 0: Tracing disabled, since +tracefile was not provided.
random min: 0x0, random max: 0xffffffffffffffff
Commencing simulation.
[    0.000000] OF: fdt: No chosen node found, continuing without
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Forcing kernel command line to: console=hvc0 earlycon=sbi
[    0.000000] Linux version 5.7.0-rc3-58539-g5f5fd87b36e2 (soc@p520) (gcc version 9.2.0 (GCC), GNU ld (GNU Binutils) 2.32) #1 SMP Sat Jul 16 09:47:13 CST 2022
[    0.000000] earlycon: sbi0 at I/O port 0x0 (options '')
[    0.000000] printk: bootconsole [sbi0] enabled
[    0.000000] initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x000000047fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x000000047fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x000000047fffffff]
[    0.000000] software IO TLB: mapped [mem 0xfbfff000-0xfffff000] (64MB)
[    0.000000] SBI specification v0.1 detected

Block design is shown in the following figure