firesim/firesim #1142: pytests failing on main@7839d4fd

timsnyder-siv, 2022-07-24:
If I checkout the main branch (currently https://github.com/firesim/firesim/commit/7839d4fda5eb360841694833c40f50e3e9bf3387)  and run pytest, there are failing tests. I expect that the pytests should be passing on main but it appears that something has snuck through causing them to fail.

I'm going to keep digging into this but wanted to file an issue so that others are aware.

timsnyder-siv, 2022-07-24:
One of the failures looks like:
```
    @wraps(func)
    def wrapper(self, *args, **kw):
        try:
            value = func(self, *args, **kw)
        except AssertionError as e:
>           raise AssertionError(e)
E           AssertionError: When calling b'main [firesim.py line 428]' the exception message does not match. Expected to match regex: 'INVALID_WORKLOAD'
E            against:
E            "[Errno 2] No such file or directory: 'config_build_recipes.yaml'"

../conda/envs/firesim/lib/python3.8/site-packages/sure/__init__.py:410: AssertionError
```

Thinking about this, it seems like it is being caused by manager support for metasim being added and wanting to read the `config_build_recipes.yaml` file when doing a "run" task.  That is new behavior of the manager added by metasim support and the `run_yamls` fixture in https://github.com/firesim/firesim/blob/7839d4fda5eb360841694833c40f50e3e9bf3387/deploy/tests/test_yaml_api.py#L169-L170 only writes `config_runtime.yaml` and `config_hwdb.yaml`.

I thought that #1076 may have caused them to start failing but looking at https://github.com/firesim/firesim/runs/6852034105?check_suite_focus=true#step:3:36 they passed. 

Rather than trying to track down where they didn't get run or started failing and were ignored, I'm going to bisect from where #1076 was merged into main to today's main and find the failing point.

timsnyder-siv, 2022-07-24:
It's weird that the CI passed on #1076 because when I checkout 0329104af77aa44af7d58218a4d4ce180a13361d where it was merged into main, the pytests fail.

timsnyder-siv, 2022-07-24:
And checking out the first parent of the merge of #1076 (previous value of main before merging it), the pytests pass.

timsnyder-siv, 2022-07-24:
Looking more into the CI run of #1076, I see it mentioning that it is running with SHA '4e5e0959b7d27e9d3f717c9558eeb926596eca3c' https://github.com/firesim/firesim/runs/6852034105?check_suite_focus=true#step:2:451

And I don't see that version existing anywhere now. I really don't understand how it was able to pass in the PR.

I'm just going to create a PR that fixes the pytests rather than try to understand this more.

timsnyder-siv, 2022-07-25:
Even with changes to the python tests, they weren't run in #1143.  Looks like something is definitely not right with the rules for skipping them.

davidbiancolin, 2022-07-25:
So that SHA is a merge commit of the PR and base branch. You'll need to pull in more refs to check it out directly. Yeah, the filtering rules are borked :(

davidbiancolin, 2022-07-27:
Resolved by #1145

closed 2022-07-27 by davidbiancolin