firesim/firesim #1171: run-basic-linux-poweroff-vitis failing because of new deps in machine-launch-script.sh

timsnyder-siv, 2022-08-15:
https://github.com/firesim/firesim/pull/1140 introduced new dependencies in `machine-launch-script.sh` and run-basic-linux-poweroff-vitis was skipped in the CI for that PR.  I'm assuming that the check was skipped because it assumes everything from machine-launch-script has been installed on the machine.  Now, it fails for later PR's that don't modify `machine-launch-script.sh` because the new deps aren't installed.  For example, https://github.com/firesim/firesim/pull/1163 in https://github.com/firesim/firesim/runs/7777886117?check_suite_focus=true#step:3:923

@abejgonzalez I'm filing this as a heads up.  If you can install `fsspec` and `s3fs` on the machine for `run-basic-linux-poweroff-vitis`, I think it will allow it to start passing

abejgonzalez, 2022-08-15:
I've updated the CI conda env. on that machine and restarted that CI run. Hopefully, that solves the issue.

timsnyder-siv, 2022-08-16:
Looks like it is failing for what is probably a bug in the #1140 code.  Can you get me `/scratch/github-actions-runners/actions-runner/_work/firesim/firesim/deploy/logs/2022-08-15--23-31-42-runworkload-VYP7IQCOPKUCD9NF.log` from `firesim1`?

timsnyder-siv, 2022-08-23:
I fixed this by reverting #1140 and I'll submit a new PR that doesn't break Vitis at some point (hopefully soon)

closed 2022-08-23 by timsnyder-siv