firesim/firesim #1172: FPGA Developer AMI - 1.11.1-40257ab5-6688-4c95-97d1-e251a40fd1fc is missing from Amazon

esromneb, 2022-08-16:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

I've tried multiple git tags
[image]

### OS Setup

```
<!-- copy/paste the output of `uname -a; lsb_release -a` below -->
```
Trying to use the AWS AMI

### Other Setup

### Current Behavior

I'm trying to follow https://docs.fires.im/en/latest/Initial-Setup/Setting-up-your-Manager-Instance.html but it appears that the 1.11.1 AMI is missing. If I reduce my search to just `40257ab5-6688-4c95-97d1-e251a40fd1fc` two other versions appear. I'm trying to find a combination of AMI version and git tag that will work but so far I've run into multiple errors. The errors I've seen
* https://github.com/firesim/firesim/issues/998
* https://github.com/firesim/firesim/issues/980
* `ld: Error: unable to disambiguate: -no-pie (did you mean --no-pie ?)`
* `ERROR: pixman >= 0.21.8 not present.`

Versions I've tried:
* `AMI 1.13.0`
* `AMI 1.12.1`

Git tags I've tried
* `1.14.1`
* `1.14.0`
* `1.13.0`
* `main` (`a4acaefe`)
* `651c9c86` (as mentioned by https://github.com/firesim/firesim/issues/998)

### Expected Behavior

I'm just looking for a simple combination of AMI and tag that will work. An updated tutorial would also be nice.

sagark, 2022-08-17:
Sorry about this, it seems that for some reason the 1.11.1 AMI was delisted recently without notification. Interestingly, it seems that any programmatic mechanism for launching instances (like used in firesim's CI) continues to be able to find the correct version, so nobody noticed :(

We will work on updating to a version that is visible in the AWS launch GUI.

In the meantime, can you try the following? I was able to launch an instance with the 1.11.1 AMI through the GUI this way:

1. Go to the marketplace page for the FPGA Developer AMI: https://aws.amazon.com/marketplace/pp/prodview-gimv3gqbpe57k
2. Click "Continue to Subscribe"
3. Click "Continue to Configuration"
4. Select 1.11.1 in the "Software version" dropdown (note that the list appears to be ordered by release date, not version number)
5. Click "Continue to Launch"
6. Select "Launch through EC2" in the "Choose Action" dropdown.
7. Click "Launch"
8. At this point, you can follow the usual documentation again starting from "4. In the Instance Type drop-down...". https://docs.fires.im/en/stable/Initial-Setup/Setting-up-your-Manager-Instance.html#launching-a-manager-instance

Let me know if this works and I will update the documentation until we can test a new AMI.

esromneb, 2022-08-18:
Sadly 1.11.1 is not in this list either. I just double checked and this is what I get:
[image]

minmin20028, 2022-08-29:
> Sadly 1.11.1 is not in this list either. I just double checked and this is what I get: [image]

Hi. 

I have the same problem.
Have you found a combination of AMI and tag that work? or working on version 1.11.2?
It would be appreciated if you could share information.

Thank you.

sagark, 2022-08-29:
We're currently working on bumping FireSim to work with AMI 1.12.1. Unfortunately, 1.11.1 was deprecated without warning and I've been traveling so did not have time to fix this. In the meantime there are two options (@esromneb and @minmin20028, you should directly go to option 2):

# Option 1: 

This will only work if you subscribed to the FPGA Developer AMI prior to 1.11.1 being deprecated. 

If this is the case, try the instructions in https://github.com/firesim/firesim/issues/1172#issuecomment-1217330446

If this works for you, you can proceed to use FireSim normally.

If you do not see 1.11.1 when following those instructions, use Option 2:

# Option 2:

Follow the usual FireSim documentation, but with two changes:

1. Select the 1.12.1 AMI
2. Switch to this branch right after you are instructed to clone firesim: https://github.com/firesim/firesim/pull/1174

This should be enough for you to get started for now; you can set up firesim normally and run simulations in the intro documentation using included AGFIs. Re-generating fpga images with the new vivado version in 1.12.1 is currently broken, which I am looking into. Once that's done, I will backport and ship a release. If you're stuck using Option 2, please subscribe to this issue and I will post an update here once we release a new version.

sagark, 2022-08-31:
We have shipped a new release that resolves this issue: https://github.com/firesim/firesim/releases/tag/1.14.2

Also, thanks for pointing out the documentation issues, we have changed the readthedocs defaults so that "latest" and "stable" always point to the newest marked version release. Switching to the latest dev docs now requires explicitly changing to the "main" version on readthedocs.

sagark, 2022-08-31:
Well... it seems like they have just de-listed AMI 1.12.1 too --- the only AMI left is 1.13.0, which doesn't have a corresponding working version of aws-fpga to use, so we're somewhat stuck at the moment...

If you subscribed to the FPGA Developer AMI before 1.12.1 was delisted, you should still be able to use FireSim using the newly released 1.14.2 branch of FireSim + using the instructions in https://github.com/firesim/firesim/issues/1172#issuecomment-1217330446 to launch a new instance using the 1.12.1 AMI (selecting 1.12.1 in the dropdown instead of 1.11.1).

davidbiancolin, 2022-09-04:
Ok following the instructions in 1.14.2 appears to work fine now. I'm not entirely sure what changed.