firesim/firesim #1192: Error when building an AFI image

nanmuxx, 2022-08-31:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

1.14.1-0-gd7a8b6c5

### OS Setup

```
<!-- copy/paste the output of `uname -a; lsb_release -a` below -->
```
Linux ip-192-168-1-216.ec2.internal 3.10.0-1160.49.1.el7.x86_64 #1 SMP Tue Nov 30 15:51:32 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
LSB Version:    :core-4.1-amd64:core-4.1-noarch
Distributor ID: CentOS
Description:    CentOS Linux release 7.9.2009 (Core)
Release:        7.9.2009
Codename:       Core

### Other Setup

I tried to build an AFI image of CVA6. And I just changed the deploy/config_build.yaml. 

build_farm:
  # managerinit replace start
  base_recipe: build-farm-recipes/aws_ec2.yaml
  # Uncomment and add args to override defaults.
  # Arg structure should be identical to the args given
  # in the base_recipe.
  #recipe_arg_overrides:
  #  <ARG>: <OVERRIDE>
  # managerinit replace end
builds_to_run:
    # this section references builds defined in config_build_recipes.yaml
    # if you add a build here, it will be built when you run buildafi

    # Unnetworked designs use a three-domain configuration
    # Tiles: 1600 MHz
    #    <Rational Crossing>
    # Uncore: 800 MHz
    #    <Async Crossing>
    # DRAM : 1000 MHz
    - firesim_cva6_singlecore_no_nic_l2_llc4mb_ddr3
agfis_to_share:
    - firesim_cva6_singlecore_no_nic_l2_llc4mb_ddr3

Then I ran a build linke:
 firesim buildbitstream

### Current Behavior

I met the error:

[localhost] out: [error] Nonzero exit code: 1
[localhost] out: [error] (Compile / runMain) Nonzero exit code: 1
[localhost] out: [error] Total time: 10 s, completed Aug 31, 2022, 10:32:20 AM
[localhost] out: make: *** [/home/centos/firesim/sim/src/main/makefrag/firesim/Makefrag:44: /home/centos/firesim/sim/generated-src/f1/FireSim-DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.CVA6Config-WithAutoILA_F90MHz_BaseF1Config/firesim.firesim.FireSim.DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.CVA6Config.fir] Error 1
[localhost] out:
Fatal error: run() received nonzero return code 2 while executing!
Requested: make  DESIGN=FireSim TARGET_CONFIG=DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.CVA6Config PLATFORM_CONFIG=WithAutoILA_F90MHz_BaseF1Config PLATFORM=f1 replace-rtl
Executed: /bin/bash -l -c "cd /home/centos/firesim/deploy/../ && export RISCV=/home/centos/firesim/target-design/chipyard/riscv-tools-install && export PATH=/home/centos/firesim/utils/fireperf:/home/centos/firesim/utils/fireperf/FlameGraph:/home/centos/firesim/target-design/chipyard/riscv-tools-install/bin:/home/centos/firesim/target-design/chipyard/software/firemarshal:/home/centos/firesim/target-design/chipyard/bin:/opt/Xilinx/Vitis/2021.1/bin:/opt/Xilinx/Vivado/2021.1/bin:/opt/conda/envs/firesim/bin:/opt/conda/condabin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/srv/git/centos-git-common:/home/centos/.local/bin:/home/centos/bin:/home/centos/firesim/deploy && export LD_LIBRARY_PATH=/home/centos/firesim/sim/lib-install/lib:/home/centos/firesim/target-design/chipyard/riscv-tools-install/lib && source sourceme-f1-manager.sh && cd sim/ && make  DESIGN=FireSim TARGET_CONFIG=DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.CVA6Config PLATFORM_CONFIG=WithAutoILA_F90MHz_BaseF1Config PLATFORM=f1 replace-rtl"
Aborting.
Fatal error.
Traceback (most recent call last):
  File "/home/centos/firesim/deploy/firesim", line 508, in <module>
    main(args)
  File "/home/centos/firesim/deploy/firesim", line 448, in main
    t['task'](t['config'](args))
  File "/home/centos/firesim/deploy/firesim", line 257, in buildbitstream
    execute(build_config.bitbuilder.replace_rtl, hosts=['localhost'])
  File "/opt/conda/envs/firesim/lib/python3.8/site-packages/fabric/tasks.py", line 377, in execute
    results[host] = _execute(
  File "/opt/conda/envs/firesim/lib/python3.8/site-packages/fabric/tasks.py", line 268, in _execute
    return task.run(*args, **kwargs)
  File "/opt/conda/envs/firesim/lib/python3.8/site-packages/fabric/tasks.py", line 168, in run
    return self.wrapped(*args, **kwargs)
  File "/home/centos/firesim/deploy/buildtools/bitbuilder.py", line 121, in replace_rtl
    run(self.build_config.make_recipe("PLATFORM=f1 replace-rtl"))
  File "/opt/conda/envs/firesim/lib/python3.8/site-packages/fabric/network.py", line 700, in host_prompting_wrapper
    return func(*args, **kwargs)
  File "/opt/conda/envs/firesim/lib/python3.8/site-packages/fabric/operations.py", line 1068, in run
    return _run_command(
  File "/opt/conda/envs/firesim/lib/python3.8/site-packages/fabric/operations.py", line 944, in _run_command
    error(message=msg, stdout=out, stderr=err)
  File "/opt/conda/envs/firesim/lib/python3.8/site-packages/fabric/utils.py", line 357, in error
    return func(message)
  File "/opt/conda/envs/firesim/lib/python3.8/site-packages/fabric/utils.py", line 65, in abort
    raise e
SystemExit: 1

Is there a problem with this? How can I solve this problem? 

### Expected Behavior

I want to generate an AGFI of CVA6.

davidbiancolin, 2022-08-31:
Can you try running: 

`make DESIGN=FireSim TARGET_CONFIG=DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.CVA6Config PLATFORM_CONFIG=WithAutoILA_F90MHz_BaseF1Config PLATFORM=f1 replace-rtl` 

in `sim/` and report back the error?

nanmuxx, 2022-09-01:
> 
Here is the error:

make[1]: *** /home/centos/firesim/sim/generators/cva6/src/main/resources/vsrc: No such file or directory.  Stop.
[error] (run-main-0) java.lang.reflect.InvocationTargetException
[error] java.lang.reflect.InvocationTargetException
[error]         at ... ()
[error]         at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:36)
[error]         at ... ()
[error]         at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
[error] Caused by: java.lang.IllegalArgumentException: requirement failed: Failed to run preprocessing step
[error]         at scala.Predef$.require(Predef.scala:281)
[error]         at cva6.CVA6CoreBlackbox.<init>(CVA6CoreBlackbox.scala:149)
[error]         at cva6.CVA6TileModuleImp.$anonfun$core$2(CVA6Tile.scala:255)
[error]         at chisel3.Module$.do_apply(Module.scala:53)
[error]         at cva6.CVA6TileModuleImp.$anonfun$core$1(CVA6Tile.scala:234)
[error]         at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
[error]         at cva6.CVA6TileModuleImp.<init>(CVA6Tile.scala:234)
[error]         at cva6.CVA6Tile.$anonfun$module$1(CVA6Tile.scala:162)
[error]         at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
[error]         at cva6.CVA6Tile.module$lzycompute(CVA6Tile.scala:162)
[error]         at cva6.CVA6Tile.module(CVA6Tile.scala:162)
[error]         at cva6.CVA6Tile.module(CVA6Tile.scala:109)
[error]         at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$3(LazyModule.scala:278)
[error]         at chisel3.Module$.do_apply(Module.scala:53)
[error]         at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$2(LazyModule.scala:278)
[error]         at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
[error]         at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$1(LazyModule.scala:278)
[error]         at scala.collection.immutable.List.flatMap(List.scala:338)
[error]         at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate(LazyModule.scala:276)
[error]         at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate$(LazyModule.scala:273)
[error]         at freechips.rocketchip.diplomacy.LazyRawModuleImp.instantiate(LazyModule.scala:344)
[error]         at freechips.rocketchip.diplomacy.LazyRawModuleImp.$anonfun$x$22$2(LazyModule.scala:357)
[error]         at chisel3.withClockAndReset$.apply(MultiClock.scala:26)
[error]         at freechips.rocketchip.diplomacy.LazyRawModuleImp.$anonfun$x$22$1(LazyModule.scala:357)
[error]         at chisel3.internal.plugin.package$.autoNameRecursivelyProduct(package.scala:48)
[error]         at freechips.rocketchip.diplomacy.LazyRawModuleImp.<init>(LazyModule.scala:356)
[error]         at freechips.rocketchip.prci.Domain$$anon$1.<init>(ClockDomain.scala:10)
[error]         at freechips.rocketchip.prci.Domain.$anonfun$module$1(ClockDomain.scala:10)
[error]         at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
[error]         at freechips.rocketchip.prci.Domain.module$lzycompute(ClockDomain.scala:10)
[error]         at freechips.rocketchip.prci.Domain.module(ClockDomain.scala:10)
[error]         at freechips.rocketchip.prci.Domain.module(ClockDomain.scala:7)
[error]         at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$3(LazyModule.scala:278)
[error]         at chisel3.Module$.do_apply(Module.scala:53)
[error]         at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$2(LazyModule.scala:278)
[error]         at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
[error]         at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$1(LazyModule.scala:278)
[error]         at scala.collection.immutable.List.flatMap(List.scala:338)
[error]         at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate(LazyModule.scala:276)
[error]         at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate$(LazyModule.scala:273)
[error]         at freechips.rocketchip.diplomacy.LazyRawModuleImp.instantiate(LazyModule.scala:344)
[error]         at freechips.rocketchip.diplomacy.LazyRawModuleImp.$anonfun$x$22$2(LazyModule.scala:357)
[error]         at chisel3.withClockAndReset$.apply(MultiClock.scala:26)
[error]         at freechips.rocketchip.diplomacy.LazyRawModuleImp.$anonfun$x$22$1(LazyModule.scala:357)
[error]         at chisel3.internal.plugin.package$.autoNameRecursivelyProduct(package.scala:48)
[error]         at freechips.rocketchip.diplomacy.LazyRawModuleImp.<init>(LazyModule.scala:356)
[error]         at freechips.rocketchip.prci.Domain$$anon$1.<init>(ClockDomain.scala:10)
[error]         at freechips.rocketchip.prci.Domain.$anonfun$module$1(ClockDomain.scala:10)
[error]         at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
[error]         at freechips.rocketchip.prci.Domain.module$lzycompute(ClockDomain.scala:10)
[error]         at freechips.rocketchip.prci.Domain.module(ClockDomain.scala:10)
[error]         at freechips.rocketchip.prci.Domain.module(ClockDomain.scala:7)
[error]         at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$3(LazyModule.scala:278)
[error]         at chisel3.Module$.do_apply(Module.scala:53)
[... cut ...]
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 18 s, completed Sep 1, 2022, 1:50:46 AM
make: *** [/home/centos/firesim/sim/src/main/makefrag/firesim/Makefrag:44: /home/centos/firesim/sim/generated-src/f1/FireSim-DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.CVA6Config-WithAutoILA_F90MHz_BaseF1Config/firesim.firesim.FireSim.DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimConfigTweaks_chipyard.CVA6Config.fir] Error 1

And the directory 'generators/' does not exist in /firesim/sim/.

davidbiancolin, 2022-09-01:
Yeah, that's a chipyard directory. This might work better if you clone Chipyard and then use FireSim as a library in there. See: https://chipyard.readthedocs.io/en/stable/Simulation/FPGA-Accelerated-Simulation.html?highlight=firesim for reference

@abejgonzalez do you have any takes on this?

abejgonzalez, 2022-09-01:
CVA6 in Chipyard works a bit weird (at least different from the traditional Chisel generators). Under the hood (during the Chisel elaboration) it runs a Makefile to remove Verilog `include` directives. See below:

https://github.com/ucb-bar/cva6-wrapper/blob/705c48a1dacc011cef9b4d021a88b4948e7f9b64/src/main/scala/CVA6CoreBlackbox.scala#L143-L149

There you can see the error message. Can you try running in `target-design/chipyard/generators/cva6/src/main/resources/vsrc/` the following command and show/understand the output?:

```bash
make default
```

There isn't enough information to know whats going on but if I was to guess, the CVA6 verilog sources are not getting checked out correctly, were modified and the Makefile doesn't like it, or something like that. It's probably worthwhile running a `git submodule status` and `git status` within the Chipyard directory (`target-design/chipyard`).

nanmuxx, 2022-09-04:
When I run `./build-setup.sh fast` in the Firesim directory, an incorrect Vivado version error has occurred, which did not occur before. 

INFO: Setting up environment variables
INFO: Changing AWS_FPGA_REPO_DIR from /home/centos/src/project_data/aws-fpga to /home/centos/firesim/platforms/f1/aws-fpga
INFO: Setting up Vivado patches if required.
INFO: Checking if patch AR71715 needs to be installed
INFO: Checking if patch AR73068 needs to be installed
INFO: Patch AR73068 not applicable for Vivado Vivado v2022.1 (64-bit).
INFO: Using Vivado v2022.1 (64-bit)
ERROR: Vivado v2022.1 (64-bit) is not supported by this HDK release.
ERROR: Supported versions are:
Vivado v2019.1.op (64-bit)
Vivado v2019.1 (64-bit)
Vivado v2019.1_AR73068 (64-bit)
Vivado v2019.1_AR73068_op (64-bit)
Vivado v2019.1_AR72668 (64-bit)
Vivado v2019.2 (64-bit)
Vivado v2019.2_AR73068_op (64-bit)
Vivado v2019.2_AR73068 (64-bit)
Vivado v2020.1 (64-bit)
Vivado v2020.2 (64-bit)
Vivado v2021.1 (64-bit)
Vivado v2021.2 (64-bit)