firesim/firesim #1231: TracerV segfaults on a bare-metal workload

vargandhi, 2022-09-25:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

1.14.2-0-g8c5416c0 

### OS Setup

Linux ip-192-168-0-95.ec2.internal 3.10.0-1160.66.1.el7.x86_64 #1 SMP Wed May 18 16:02:34 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
LSB Version:    :core-4.1-amd64:core-4.1-noarch
Distributor ID: CentOS
Description:    CentOS Linux release 7.9.2009 (Core)
Release:        7.9.2009
Codename:       Core

### Other Setup

I'm running firesim as a submodule of chipyard.

### Current Behavior

I’m trying to profile a sample bare-metal workload defined in bare.yaml (firemarshal/test/) using tracerV on a quad-core-rocket config. The simulation runs fine without tracerV enabled, and I see “Hello World” in the uartlog generated. However, when I enable tracerV, the simulation terminates with a segfault:

Script started on Fri 23 Sep 2022 12:01:36 AM UTC
AFI ID for Slot  0: agfi-0c1b4d7cf927b4df8^M
AFI PCI  Vendor ID: 0x1d0f, Device ID 0xf000^M
Using xdma write queue: /dev/xdma0_h2c_0^M
Using xdma read queue: /dev/xdma0_c2h_0^M
UART0 is here (stdin/stdout).^M
command line for program 0. argc=57:^M
+permissive +mm_relaxFunctionalModel_0=0 +mm_openPagePolicy_0=1 +mm_backendLatency_0=2 +mm_schedulerWindowSize_0=8 +mm_transactionQueueDepth_0=8 +mm_dramTimings_tAL_0=0 +mm_dramTimings_tCAS_0=14 +mm_dramTimings_tCMD_0=1 +mm_dramTimings_tCWD_0=10 +mm_dramTimings_tCCD_0=4 +mm_dramTimings_tFAW_0=25 +mm_dramTimings_tRAS_0=33 +mm_dramTimings_tREFI_0=7800 +mm_dramTimings_tRC_0=47 +mm_dramTimings_tRCD_0=14 +mm_dramTimings_tRFC_0=160 +mm_dramTimings_tRRD_0=8 +mm_dramTimings_tRP_0=14 +mm_dramTimings_tRTP_0=8 +mm_dramTimings_tRTRS_0=2 +mm_dramTimings_tWR_0=15 +mm_dramTimings_tWTR_0=8 +mm_rowAddr_offset_0=18 +mm_rowAddr_mask_0=65535 +mm_rankAddr_offset_0=16 +mm_rankAddr_mask_0=3 +mm_bankAddr_offset_0=13 +mm_bankAddr_mask_0=7 +mm_llc_wayBits_0=3 +mm_llc_setBits_0=12 +mm_llc_blockBits_0=7 +mm_llc_activeMSHRs_0=8 +slotid=0 +profile-interval=-1 +macaddr0=00:12:6D:00:00:02 +blkdev0=bare-mem0-dummy.rootfs +niclog0=niclog0 +blkdev-log0=blkdev-log0 +tracefile=TRACEFILE +trace-select=0 +trace-start=0 +trace-end=-1 +trace-output-format=0 +dwarf-file-name=bare-mem0-hello-dwarf +autocounter-readrate=100 +autocounter-filename-base=AUTOCOUNTERFILE +drj_dtb=bare-mem0-hello.dtb +drj_bin=bare-mem0-hello +drj_rom=bare-mem0-hello.rom +print-start=0 +print-end=-1 +linklatency0=6405 +netbw0=200 +shmemportname0=0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 +permissive-off bare-mem0-hello ^M
using link latency: 6405 cycles^M
using netbw: 200^M
using netburst: 8^M
Using non-slot-id associated shmemportname:^M
opening/creating shmem region^M
/port_nts0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000_0^M
Using non-slot-id associated shmemportname:^M
opening/creating shmem region^M
/port_stn0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000_0^M
Using non-slot-id associated shmemportname:^M
opening/creating shmem region^M
/port_nts0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000_1^M
Using non-slot-id associated shmemportname:^M
opening/creating shmem region^M
/port_stn0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000_1^M
random min: 0x0, random max: 0xffffffffffffffff^M
TracerV: No trigger selected. Trigger enabled from 0 to 18446744073709551615 cycles^M
TracerV: No trigger selected. Trigger enabled from 0 to 18446744073709551615 cycles^M
TracerV: No trigger selected. Trigger enabled from 0 to 18446744073709551615 cycles^M
TracerV: No trigger selected. Trigger enabled from 0 to 18446744073709551615 cycles^M
Commencing simulation.^M
bash: line 1: 17955 Segmentation fault      sudo ./FireSim-f1 +permissive +mm_relaxFunctionalModel_0=0 +mm_openPagePolicy_0=1 +mm_backendLatency_0=2 +mm_schedulerWindowSize_0=8 +mm_transactionQueueDepth_0=8 +mm_dramTimings_tAL_0=0 +mm_dramTimings_tCAS_0=14 +mm_dramTimings_tCMD_0=1 +mm_dramTimings_tCWD_0=10 +mm_dramTimings_tCCD_0=4 +mm_dramTimings_tFAW_0=25 +mm_dramTimings_tRAS_0=33 +mm_dramTimings_tREFI_0=7800 +mm_dramTimings_tRC_0=47 +mm_dramTimings_tRCD_0=14 +mm_dramTimings_tRFC_0=160 +mm_dramTimings_tRRD_0=8 +mm_dramTimings_tRP_0=14 +mm_dramTimings_tRTP_0=8 +mm_dramTimings_tRTRS_0=2 +mm_dramTimings_tWR_0=15 +mm_dramTimings_tWTR_0=8 +mm_rowAddr_offset_0=18 +mm_rowAddr_mask_0=65535 +mm_rankAddr_offset_0=16 +mm_rankAddr_mask_0=3 +mm_bankAddr_offset_0=13 +mm_bankAddr_mask_0=7 +mm_llc_wayBits_0=3 +mm_llc_setBits_0=12 +mm_llc_blockBits_0=7 +mm_llc_activeMSHRs_0=8 +slotid=0 +profile-interval=-1 +macaddr0=00:12:6D:00:00:02 +blkdev0=bare-mem0-dummy.rootfs +niclog0=niclog0 +blkdev-log0=blkdev-log0 +tracefile=TRACEFILE +trace-select=0 +trace-start=0 +trace-end=-1 +trace-output-format=0 +dwarf-file-name=bare-mem0-hello-dwarf +autocounter-readrate=100 +autocounter-filename-base=AUTOCOUNTERFILE +drj_dtb=bare-mem0-hello.dtb +drj_bin=bare-mem0-hello +drj_rom=bare-mem0-hello.rom +print-start=0 +print-end=-1 +linklatency0=6405 +netbw0=200 +shmemportname0=0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 +permissive-off +prog0=bare-mem0-hello^M

### Expected Behavior

The bare-metal workload should run and generate a uartlog with the "Hello World" output, i.e., similar to the one generated when tracerV is not enabled.

Script started on Thu 22 Sep 2022 11:19:32 PM UTC
AFI ID for Slot  0: agfi-0c1b4d7cf927b4df8^M
AFI PCI  Vendor ID: 0x1d0f, Device ID 0xf000^M
Using xdma write queue: /dev/xdma0_h2c_0^M
Using xdma read queue: /dev/xdma0_c2h_0^M
UART0 is here (stdin/stdout).^M
command line for program 0. argc=56:^M
+permissive +mm_relaxFunctionalModel_0=0 +mm_openPagePolicy_0=1 +mm_backendLatency_0=2 +mm_schedulerWindowSize_0=8 +mm_transactionQueueDepth_0=8 +mm_dramTimings_tAL_0=0 +mm_dramTimings_tCAS_0=14 +mm_dramTimings_tCMD_0=1 +mm_dramTimings_tCWD_0=10 +mm_dramTimings_tCCD_0=4 +mm_dramTimings_tFAW_0=25 +mm_dramTimings_tRAS_0=33 +mm_dramTimings_tREFI_0=7800 +mm_dramTimings_tRC_0=47 +mm_dramTimings_tRCD_0=14 +mm_dramTimings_tRFC_0=160 +mm_dramTimings_tRRD_0=8 +mm_dramTimings_tRP_0=14 +mm_dramTimings_tRTP_0=8 +mm_dramTimings_tRTRS_0=2 +mm_dramTimings_tWR_0=15 +mm_dramTimings_tWTR_0=8 +mm_rowAddr_offset_0=18 +mm_rowAddr_mask_0=65535 +mm_rankAddr_offset_0=16 +mm_rankAddr_mask_0=3 +mm_bankAddr_offset_0=13 +mm_bankAddr_mask_0=7 +mm_llc_wayBits_0=3 +mm_llc_setBits_0=12 +mm_llc_blockBits_0=7 +mm_llc_activeMSHRs_0=8 +slotid=0 +profile-interval=-1 +macaddr0=00:12:6D:00:00:02 +blkdev0=bare-mem0-dummy.rootfs +niclog0=niclog0 +blkdev-log0=blkdev-log0 +trace-select=1 +trace-start=0 +trace-end=-1 +trace-output-format=0 +dwarf-file-name=bare-mem0-hello-dwarf +autocounter-readrate=0 +autocounter-filename-base=AUTOCOUNTERFILE +drj_dtb=bare-mem0-hello.dtb +drj_bin=bare-mem0-hello +drj_rom=bare-mem0-hello.rom +print-start=0 +print-end=-1 +linklatency0=6405 +netbw0=200 +shmemportname0=0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 +permissive-off bare-mem0-hello ^M
[... 125 lines cut ...]
    terminate_on_completion: no
    suffix_tag: null

host_debug:
    # When enabled (=yes), Zeros-out FPGA-attached DRAM before simulations
    # begin (takes 2-5 minutes).
    # In general, this is not required to produce deterministic simulations on
    # target machines running linux. Enable if you observe simulation non-determinism.
    zero_out_dram: no
    # If disable_synth_asserts: no, simulation will print assertion message and
    # terminate simulation if synthesized assertion fires.
    # If disable_synth_asserts: yes, simulation ignores assertion firing and
    # continues simulation.
    disable_synth_asserts: no

# DOCREF START: Synthesized Prints
synth_print:
    # Start and end cycles for outputting synthesized prints.
    # They are given in terms of the base clock and will be converted
    # for each clock domain.
    start: 0
    end: -1
    # When enabled (=yes), prefix print output with the target cycle at which the print was triggered
    cycle_prefix: yes
# DOCREF END: Synthesized Prints

davidbiancolin, 2022-09-27:
Hey Varan, could you confirm this still happens using one of the pre-compiled RISC-V baremetal tests. I'd try a longer one such as mm, which you can find here:
 $RISCV/riscv64-unknown-elf/share/riscv-tests/benchmarks/mm.riscv

vargandhi, 2022-09-27:
Hi David,

Yes, I can confirm that tracerV segfaults on mm.riscv. Whereas mm.riscv works fine when I disable tracing from my config_runtime.yaml

Also, I’m running these workloads on a quad-core medium Boom config. Not sure if that would make a difference though...

Here’s the uartlog for the two runs —

TracerV enabled 

Script started on Tue 27 Sep 2022 09:01:10 PM UTC
AFI ID for Slot  0: agfi-0568d760705e5f378^M
AFI PCI  Vendor ID: 0x1d0f, Device ID 0xf000^M
Using xdma write queue: /dev/xdma0_h2c_0^M
Using xdma read queue: /dev/xdma0_c2h_0^M
UART0 is here (stdin/stdout).^M
command line for program 0. argc=57:^M
+permissive +mm_relaxFunctionalModel_0=0 +mm_openPagePolicy_0=1 +mm_backendLatency_0=2 +mm_schedulerWindowSize_0=8 +mm_transactionQueueDepth_0=8 +mm_dramTimings_tAL_0=0 +mm_dramTimings_tCAS_0=14 +mm_dramTimings_tCMD_0=1 +mm_dramTimings_tCWD_0=10 +mm_dramTimings_tCCD_0=4 +mm_dramTimings_tFAW_0=25 +mm_dramTimings_tRAS_0=33 +mm_dramTimings_tREFI_0=7800 +mm_dramTimings_tRC_0=47 +mm_dramTimings_tRCD_0=14 +mm_dramTimings_tRFC_0=160 +mm_dramTimings_tRRD_0=8 +mm_dramTimings_tRP_0=14 +mm_dramTimings_tRTP_0=8 +mm_dramTimings_tRTRS_0=2 +mm_dramTimings_tWR_0=15 +mm_dramTimings_tWTR_0=8 +mm_rowAddr_offset_0=18 +mm_rowAddr_mask_0=65535 +mm_rankAddr_offset_0=16 +mm_rankAddr_mask_0=3 +mm_bankAddr_offset_0=13 +mm_bankAddr_mask_0=7 +mm_llc_wayBits_0=3 +mm_llc_setBits_0=12 +mm_llc_blockBits_0=7 +mm_llc_activeMSHRs_0=8 +slotid=0 +profile-interval=-1 +macaddr0=00:12:6D:00:00:02 +blkdev0=riscv-mm0-dummy.rootfs +niclog0=niclog0 +blkdev-log0=blkdev-log0 +tracefile=TRACEFILE +trace-select=0 +trace-start=0 +trace-end=-1 +trace-output-format=0 +dwarf-file-name=riscv-mm0-mm.riscv-dwarf +autocounter-readrate=0 +autocounter-filename-base=AUTOCOUNTERFILE +drj_dtb=riscv-mm0-mm.riscv.dtb +drj_bin=riscv-mm0-mm.riscv +drj_rom=riscv-mm0-mm.riscv.rom +print-start=0 +print-end=-1 +linklatency0=6405 +netbw0=200 +shmemportname0=0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 +permissive-off riscv-mm0-mm.riscv ^M
using link latency: 6405 cycles^M
using netbw: 200^M
using netburst: 8^M
Using non-slot-id associated shmemportname:^M
opening/creating shmem region^M
/port_nts0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000_0^M
Using non-slot-id associated shmemportname:^M
opening/creating shmem region^M
/port_stn0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000_0^M
Using non-slot-id associated shmemportname:^M
opening/creating shmem region^M
/port_nts0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000_1^M
Using non-slot-id associated shmemportname:^M
opening/creating shmem region^M
/port_stn0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000_1^M
random min: 0x0, random max: 0xffffffffffffffff^M
TracerV: No trigger selected. Trigger enabled from 0 to 18446744073709551615 cycles^M
TracerV: No trigger selected. Trigger enabled from 0 to 18446744073709551615 cycles^M
TracerV: No trigger selected. Trigger enabled from 0 to 18446744073709551615 cycles^M
TracerV: No trigger selected. Trigger enabled from 0 to 18446744073709551615 cycles^M
Commencing simulation.^M
bash: line 1: 29297 Segmentation fault      sudo ./FireSim-f1 +permissive +mm_relaxFunctionalModel_0=0 +mm_openPagePolicy_0=1 +mm_backendLatency_0=2 +mm_schedulerWindowSize_0=8 +mm_transactionQueueDepth_0=8 +mm_dramTimings_tAL_0=0 +mm_dramTimings_tCAS_0=14 +mm_dramTimings_tCMD_0=1 +mm_dramTimings_tCWD_0=10 +mm_dramTimings_tCCD_0=4 +mm_dramTimings_tFAW_0=25 +mm_dramTimings_tRAS_0=33 +mm_dramTimings_tREFI_0=7800 +mm_dramTimings_tRC_0=47 +mm_dramTimings_tRCD_0=14 +mm_dramTimings_tRFC_0=160 +mm_dramTimings_tRRD_0=8 +mm_dramTimings_tRP_0=14 +mm_dramTimings_tRTP_0=8 +mm_dramTimings_tRTRS_0=2 +mm_dramTimings_tWR_0=15 +mm_dramTimings_tWTR_0=8 +mm_rowAddr_offset_0=18 +mm_rowAddr_mask_0=65535 +mm_rankAddr_offset_0=16 +mm_rankAddr_mask_0=3 +mm_bankAddr_offset_0=13 +mm_bankAddr_mask_0=7 +mm_llc_wayBits_0=3 +mm_llc_setBits_0=12 +mm_llc_blockBits_0=7 +mm_llc_activeMSHRs_0=8 +slotid=0 +profile-interval=-1 +macaddr0=00:12:6D:00:00:02 +blkdev0=riscv-mm0-dummy.rootfs +niclog0=niclog0 +blkdev-log0=blkdev-log0 +tracefile=TRACEFILE +trace-select=0 +trace-start=0 +trace-end=-1 +trace-output-format=0 +dwarf-file-name=riscv-mm0-mm.riscv-dwarf +autocounter-readrate=0 +autocounter-filename-base=AUTOCOUNTERFILE +drj_dtb=riscv-mm0-mm.riscv.dtb +drj_bin=riscv-mm0-mm.riscv +drj_rom=riscv-mm0-mm.riscv.rom +print-start=0 +print-end=-1 +linklatency0=6405 +netbw0=200 +shmemportname0=0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 +permissive-off +prog0=riscv-mm0-mm.riscv^M

Script done on Tue 27 Sep 2022 09:01:10 PM UTC

————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————

TracerV NOT enabled 

Script started on Tue 27 Sep 2022 08:54:00 PM UTC
AFI ID for Slot  0: agfi-0568d760705e5f378^M
AFI PCI  Vendor ID: 0x1d0f, Device ID 0xf000^M
Using xdma write queue: /dev/xdma0_h2c_0^M
Using xdma read queue: /dev/xdma0_c2h_0^M
UART0 is here (stdin/stdout).^M
command line for program 0. argc=56:^M
[... cut ...]
C0: reg block 4x5x6, cache block 24x25x24^M
C0: 24688 instructions^M
C0: 16568 cycles^M
C0: 28800 flops^M
C0: 1738 Mflops @ 1 GHz^M
^M
Simulation complete.^M
*** PASSED *** after 194462207 cycles^M
^M
Emulation Performance Summary^M
------------------------------^M
Wallclock Time Elapsed: 4.7 s^M
Host Frequency: 64.999 MHz^M
Target Cycles Emulated: 194462207^M
Effective Target Frequency: 41.136 MHz^M
FMR: 1.58^M
Note: The latter three figures are based on the fastest target clock.^M

Script done on Tue 27 Sep 2022 08:54:05 PM UTC
~

vargandhi, 2022-09-30:
Hi David,

I was wondering if you got a chance to look into it. Is there anything I could try on my side to debug this issue?

Best,
Varun

davidbiancolin, 2022-10-04:
@vargandhi can you try cherry-picking the change in #1249? Let me know if it helps.

vargandhi, 2022-10-05:
@davidbiancolin -- I'm using firesim as a library in chipyard and am not able to finish the manager instance setup after modifying the firesim source code. I keep getting the following error:
Traceback (most recent call last):
  File "/home/centos/chipyard/sims/firesim/deploy/firesim", line 28, in <module>
    from runtools.runtime_config import RuntimeConfig
  File "/home/centos/chipyard/sims/firesim/deploy/runtools/runtime_config.py", line 17, in <module>
    from runtools.firesim_topology_with_passes import FireSimTopologyWithPasses
  File "/home/centos/chipyard/sims/firesim/deploy/runtools/firesim_topology_with_passes.py", line 11, in <module>
    from colorama import Fore, Style # type: ignore
ModuleNotFoundError: No module named 'colorama'

Although, at this point, I'm not sure if this is an issue on the firesim side or chipyard. I'm building the chipyard repo with branch 1.7.1 but not sure if it's still the most recent stable version. The stable version of the chipyard documentation is missing this info.

vargandhi, 2022-10-06:
@davidbiancolin -- I successfully ran tracerV after making the suggested change. Thanks!

closed 2022-10-07 by davidbiancolin