firesim/firesim #1293: [on-premise] Alveo U250 timing error `clk_kernel_01_unbuffered_net`

caizixian, 2022-11-07:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

`1.15.0-43-g8176b657` (checked out by Chipyard 1.8.1)

### OS Setup

```console
Linux alveo 5.15.0-52-generic #58~20.04.1-Ubuntu SMP Thu Oct 13 13:09:46 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.5 LTS
Release:        20.04
Codename:       focal
```

### Other Setup

I tried to follow the documentation here https://docs.fires.im/en/1.15.0/Advanced-Usage/Vitis.html

However, Xilinx no longer provides the old platform files, see, e.g. https://support.xilinx.com/s/question/0D52E00007FScTdSAL/where-to-find-old-target-platform-for-alveo-u250-eg-20202-version-or-should-i-always-update-to-the-latest-xrtplatform?language=en_US and https://support.xilinx.com/s/question/0D52E00007E9KIrSAN/um-how-to-get-u250-gen3x16-xdma-31-platform?language=en_US

So instead, I used the following.
```
Vitis/Vivado: 2022.2
XRT: `202210.2.13.479-0-g5e92a513c`
Deployment platform: `xilinx-u250-gen3x16-xdma_2022.1_2022_0415_2123-all.deb.tar.gz`
Development platform: `xilinx-u250-gen3x16-xdma-4.1-202210-1-dev_1-3512975_all.deb`
```

I changed the `config_*.yaml` files under `sims/firesim/deploy` per the documentation. FireSim is patched with the below diff.
```diff
diff --git a/deploy/buildtools/bitbuilder.py b/deploy/buildtools/bitbuilder.py
index a6e7363d..5c012486 100644
--- a/deploy/buildtools/bitbuilder.py
+++ b/deploy/buildtools/bitbuilder.py
@@ -501,7 +501,7 @@ class VitisBitBuilder(BitBuilder):
             return False
 
         hwdb_entry_name = self.build_config.name
-        xclbin_path = cl_dir + "/bitstream/build_dir.xilinx_u250_gen3x16_xdma_3_1_202020_1/firesim.xclbin"
+        xclbin_path = cl_dir + "/bitstream/build_dir.xilinx_u250_gen3x16_xdma_4_1_202210_1/firesim.xclbin"
 
         results_build_dir = """{}/""".format(local_results_dir)
 
diff --git a/platforms/vitis/cl_firesim/Makefile b/platforms/vitis/cl_firesim/Makefile
index 00134ff2..14df4815 100644
--- a/platforms/vitis/cl_firesim/Makefile
+++ b/platforms/vitis/cl_firesim/Makefile
@@ -33,7 +33,7 @@ run-sim:   TARGET = hw_emu
 run-sim:   VPP_TARGET_SPECIFIC_ARGS = --target $(TARGET) --config $(buildroot)/simulation.cfg
 
 # current support on firesim1
-DEVICE ?= xilinx_u250_gen3x16_xdma_3_1_202020_1
+DEVICE ?= xilinx_u250_gen3x16_xdma_4_1_202210_1
 
 #   device2xsa - create a filesystem friendly name from device name
 #   $(1) - full name of device
diff --git a/sim/target-agnostic.mk b/sim/target-agnostic.mk
index 2a524618..5cc214c4 100644
--- a/sim/target-agnostic.mk
+++ b/sim/target-agnostic.mk
@@ -191,7 +191,7 @@ $(f1): $(header) $(DRIVER_CC) $(DRIVER_H) $(midas_cc) $(midas_h)
 $(vitis): export CXXFLAGS := $(CXXFLAGS) $(common_cxx_flags) $(DRIVER_CXXOPTS) \
        -idirafter ${CONDA_PREFIX}/include -idirafter /usr/include -idirafter $(XILINX_XRT)/include
 $(vitis): export LDFLAGS := $(LDFLAGS) $(common_ld_flags) -Wl,-rpath='$$$$ORIGIN' \
-       -L${CONDA_PREFIX}/lib -Wl,-rpath-link=/usr/lib/x86_64-linux-gnu -L$(XILINX_XRT)/lib -luuid -lxrt_coreutil
+       -L${CONDA_PREFIX}/lib -Wl,-rpath-link=/usr/lib/x86_64-linux-gnu -L/usr/lib/x86_64-linux-gnu -L$(XILINX_XRT)/lib -luuid -lxrt_coreutil
 
 
 # Compile Driver
```

The `-L/usr/lib/x86_64-linux-gnu` linker flag is to work around the undefined references to `_dl_catch_error@GLIBC_PRIVATE` and `_dl_signal_error@GLIBC_PRIVATE` (the gcc in conda doesn't seem to search the OS library folders).

### Current Behavior

I have the following error when building bitstream.
```console
2022-11-07 04:24:29,084 [flush       ] [DEBUG]  [localhost] out: ===>The following messages were generated while  Compiling (bitstream) accelerator binary: firesim.link Log file: /home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log  :
2022-11-07 04:24:29,084 [flush       ] [DEBUG]  [localhost] out: ERROR: [VPL 101-2] design did not meet timing - Design did not meet timing. One or more unscalable system clocks did not meet their required target frequency. For all system clocks, this design is using 0 nanoseconds as the threshold worst negative slack (WNS) value. List of system clocks with timing failure:
2022-11-07 04:24:29,084 [flush       ] [DEBUG]  [localhost] out: system clock: clk_kernel_01_unbuffered_net; slack: -0.883 ns
2022-11-07 04:24:29,084 [flush       ] [DEBUG]  [localhost] out: ERROR: [VPL 101-3] sourcing script /home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/scripts/impl_1/_full_write_bitstream_pre.tcl failed
2022-11-07 04:24:29,268 [flush       ] [DEBUG]  [localhost] out: ERROR: [VPL 60-773] In '/home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/vivado.log', caught Tcl error:  problem implementing dynamic region, impl_1: write_bitstream ERROR, please look at the run log file '/home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
2022-11-07 04:24:29,346 [flush       ] [DEBUG]  [localhost] out: WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
2022-11-07 04:24:29,346 [flush       ] [DEBUG]  [localhost] out: ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, impl_1: write_bitstream ERROR, please look at the run log file '/home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
2022-11-07 04:24:29,346 [flush       ] [DEBUG]  [localhost] out: ERROR: [VPL 60-1328] Vpl run 'vpl' failed
2022-11-07 04:24:29,475 [flush       ] [DEBUG]  [localhost] out: ERROR: [VPL 60-806] Failed to finish platform linker
2022-11-07 04:24:29,485 [flush       ] [DEBUG]  [localhost] out: INFO: [v++ 60-1442] [04:24:29] Run run_link: Step vpl: Failed
2022-11-07 04:24:29,487 [flush       ] [DEBUG]  [localhost] out: Time (s): cpu = 00:00:31 ; elapsed = 01:22:38 . Memory (MB): peak = 456.906 ; gain = 0.000 ; free physical = 49081 ; free virtual = 66643
2022-11-07 04:24:29,490 [flush       ] [DEBUG]  [localhost] out: ERROR: [v++ 60-661] v++ link run 'run_link' failed
2022-11-07 04:24:29,490 [flush       ] [DEBUG]  [localhost] out: ERROR: [v++ 60-626] Kernel link failed to complete
2022-11-07 04:24:29,490 [flush       ] [DEBUG]  [localhost] out: ERROR: [v++ 60-703] Failed to finish linking
2022-11-07 04:24:29,523 [flush       ] [DEBUG]  [localhost] out: INFO: [v++ 60-1653] Closing dispatch client.
2022-11-07 04:24:39,535 [flush       ] [DEBUG]  [localhost] out: make: *** [Makefile:125: /home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/firesim.link.xclbin] Error 1
2022-11-07 04:24:39,537 [flush       ] [DEBUG]  [localhost] out: rm /home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/firesim.xo
2022-11-07 04:24:39,540 [flush       ] [DEBUG]  [localhost] out: 
2022-11-07 04:24:39,542 [flush       ] [INFO ]  Warning: run() received nonzero return code 2 while executing '/home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/build-bitstream.sh /home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig'!
```

### Expected Behavior

I'm not sure whether it is meant to build considering I'm not using the correct Vitis/XRT/BSP suggested by the documentation. But considering the old platforms are no longer provided by Xilinx, it would make sense to try the newer ones.

### Other Information

Currently trying to set `PLATFORM_CONFIG` to `F65MHz_BaseVitisConfig` and see whether it helps with the timing error.

ncppd, 2023-01-19:
I am facing the same issue, I tried to change the default frequency but the error persists. Do you have any updates on this?

abejgonzalez, 2023-01-20:
> ### Other Setup
> I tried to follow the documentation here https://docs.fires.im/en/1.15.0/Advanced-Usage/Vitis.html

That documentation needs to be updated. Right now, you can run the local FPGA setup with any Vitis/Vivado version >= 2021.1. Ideally, if you use 2022.x+ you can avoid the use of checking out the weird XRT commit and building it locally.

> However, Xilinx no longer provides the old platform files, see, e.g. [...]

This should be fine.

> I changed the `config_*.yaml` files under `sims/firesim/deploy` per the documentation. FireSim is patched with the below diff.
>
> ```diff
> diff --git a/deploy/buildtools/bitbuilder.py [...]

In FireSim `main` this should be fixed and you should be able to change the FPGA device through the `*.yaml` files: https://github.com/firesim/firesim/blob/e5b72dbf9655bfea67f11bf22c3b7dcc5a2a6500/deploy/bit-builder-recipes/vitis.yaml#L18. Then you shouldn't have to change anything else / patch the `platforms/vitis` area.

> diff --git a/platforms/vitis/cl_firesim/Makefile b/platforms/vitis/cl_firesim/Makefile
> index 00134ff2..14df4815 100644
> --- a/platforms/vitis/cl_firesim/Makefile
> +++ [...]

Yes this is an issue with Ubuntu 20 v.s. 18. I fixed this in the following PR: https://github.com/firesim/firesim/pull/1344. I instead linked in `libdl` to fix the issue (which I think is slightly nicer than having a OS-specific folder).

> ### Current Behavior
> I have the following error when building bitstream.
> 
> ```
> 2022-11-07 04:24:29,084 [flush       ] [DEBUG]  [localhost] out: ===>The following messages were generated while  Compiling (bitstream) accelerator binary: firesim.link Log file: /home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log  :
> 2022-11-07 04:24:29,084 [flush       ] [DEBUG]  [localhost] out: ERROR: [VPL 101-2] design did not meet timing - Design did not meet timing. One or more unscalable system clocks did not meet their required target frequency. For all system clocks, this design is using 0 nanoseconds as the threshold worst negative slack (WNS) value. List of system clocks with timing failure:
> 2022-11-07 04:24:29,084 [flush       ] [DEBUG]  [localhost] out: system clock: clk_kernel_01_unbuffered_net; slack: -0.883 ns
> 2022-11-07 04:24:29,084 [flush       ] [DEBUG]  [localhost] out: ERROR: [VPL 101-3] sourcing script /home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/scripts/impl_1/_full_write_bitstream_pre.tcl failed
> 2022-11-07 04:24:29,268 [flush       ] [DEBUG]  [localhost] out: ERROR: [VPL 60-773] In '/home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/vivado.log', caught Tcl error:  problem implementing dynamic region, impl_1: write_bitstream ERROR, please look at the run log file '/home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
> 2022-11-07 04:24:29,346 [flush       ] [DEBUG]  [localhost] out: WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
> 2022-11-07 04:24:29,346 [flush       ] [DEBUG]  [localhost] out: ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, impl_1: write_bitstream ERROR, please look at the run log file '/home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
> 2022-11-07 04:24:29,346 [flush       ] [DEBUG]  [localhost] out: ERROR: [VPL 60-1328] Vpl run 'vpl' failed
> 2022-11-07 04:24:29,475 [flush       ] [DEBUG]  [localhost] out: ERROR: [VPL 60-806] Failed to finish platform linker
> 2022-11-07 04:24:29,485 [flush       ] [DEBUG]  [localhost] out: INFO: [v++ 60-1442] [04:24:29] Run run_link: Step vpl: Failed
> 2022-11-07 04:24:29,487 [flush       ] [DEBUG]  [localhost] out: Time (s): cpu = 00:00:31 ; elapsed = 01:22:38 . Memory (MB): peak = 456.906 ; gain = 0.000 ; free physical = 49081 ; free virtual = 66643
> 2022-11-07 04:24:29,490 [flush       ] [DEBUG]  [localhost] out: ERROR: [v++ 60-661] v++ link run 'run_link' failed
> 2022-11-07 04:24:29,490 [flush       ] [DEBUG]  [localhost] out: ERROR: [v++ 60-626] Kernel link failed to complete
> 2022-11-07 04:24:29,490 [flush       ] [DEBUG]  [localhost] out: ERROR: [v++ 60-703] Failed to finish linking
> 2022-11-07 04:24:29,523 [flush       ] [DEBUG]  [localhost] out: INFO: [v++ 60-1653] Closing dispatch client.
> 2022-11-07 04:24:39,535 [flush       ] [DEBUG]  [localhost] out: make: *** [Makefile:125: /home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/firesim.link.xclbin] Error 1
> 2022-11-07 04:24:39,537 [flush       ] [DEBUG]  [localhost] out: rm /home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/firesim.xo
> 2022-11-07 04:24:39,540 [flush       ] [DEBUG]  [localhost] out: 
[... cut ...]
As for the timing error, I haven't seen that so I'm not sure I can help. I have been able to run bitstreams of larger/more complicated designs recently (2022.2 Vitis) so I'm not sure what is the issue. I've opened up a PR to address the documentation: https://github.com/firesim/firesim/pull/1397. I would try checking out the `main` branch of the repo (the last commit I tried was https://github.com/firesim/firesim/commit/6652b02f563d676c772cd6a24490911552013a4c if you potentially want to be safer) and following along with the docs.

caizixian, 2023-01-26:
The same error still persists with 6652b02f563d676c772cd6a24490911552013a4c

```console
[localhost] out: [14:59:56] Phase 12 Resolve XTalk
[localhost] out: [14:59:56] Phase 13 Post Router Timing
[localhost] out: [15:00:26] Phase 14 Physical Synthesis in Router
[localhost] out: [15:00:26] Phase 14.1 Physical Synthesis Initialization
[localhost] out: [15:01:56] Phase 14.2 Critical Path Optimization
[localhost] out: [15:01:56] Phase 15 Route finalize
[localhost] out: [15:02:26] Finished 5th of 6 tasks (FPGA routing). Elapsed time: 00h 31m 05s
[localhost] out:
[localhost] out: [15:02:26] Starting bitstream generation..
[localhost] out: Starting optional post-route physical design optimization.
[localhost] out: [15:13:58] Phase 1 Physical Synthesis Initialization
[localhost] out: [15:14:58] Phase 2 SLL Register Hold Fix Optimization
[localhost] out: [15:14:58] Phase 3 Critical Path Optimization
[localhost] out: Finished optional post-route physical design optimization.
[localhost] out: [15:17:12] Run vpl: Step impl: Failed
[localhost] out: [15:17:12] Run vpl: FINISHED. Run Status: impl ERROR
[localhost] out:
[localhost] out: ===>The following messages were generated while  Compiling (bitstream) accelerator binary: firesim.link Log file: /home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log  :
[localhost] out: ERROR: [VPL 101-2] design did not meet timing - Design did not meet timing. One or more unscalable system clocks did not meet their required target frequency. For all system clocks, this design is using 0 nanoseconds as the threshold worst negative slack (WNS) value. List of system clocks with timing failure:
[localhost] out: system clock: clk_kernel_01_unbuffered_net; slack: -0.889 ns
[localhost] out: ERROR: [VPL 101-3] sourcing script /home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/scripts/impl_1/_full_write_bitstream_pre.tcl failed
[localhost] out: ERROR: [VPL 60-773] In '/home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/vivado.log', caught Tcl error:  problem implementing dynamic region, impl_1: write_bitstream ERROR, please look at the run log file '/home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
[localhost] out: WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
[localhost] out: ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, impl_1: write_bitstream ERROR, please look at the run log file '/home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
[localhost] out: ERROR: [VPL 60-1328] Vpl run 'vpl' failed
[localhost] out: ERROR: [VPL 60-806] Failed to finish platform linker
[localhost] out: INFO: [v++ 60-1442] [15:17:12] Run run_link: Step vpl: Failed
[localhost] out: Time (s): cpu = 00:00:45 ; elapsed = 05:32:05 . Memory (MB): peak = 445.172 ; gain = 0.000 ; free physical = 36324 ; free virtual = 54589
[localhost] out: ERROR: [v++ 60-661] v++ link run 'run_link' failed
[localhost] out: ERROR: [v++ 60-626] Kernel link failed to complete
[localhost] out: ERROR: [v++ 60-703] Failed to finish linking
[localhost] out: INFO: [v++ 60-1653] Closing dispatch client.
[localhost] out: make: *** [Makefile:155: /home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/firesim.link.xclbin] Error 1
[localhost] out: rm /home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/_x.xilinx_u250_gen3x16_xdma_4_1_202210_1/firesim.xo
[localhost] out: make: Leaving directory '/home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig'
[localhost] out:
Warning: run() received nonzero return code 2 while executing '/home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/build-bitstream.sh --build_dir /home/zixianc/firesim_build/platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig --device xilinx_u250_gen3x16_xdma_4_1_202210_1 --frequency 140 --strategy TIMING'!
FireSim Vitis FPGA Build Failed
Your FPGA build failed for triplet: FireSim-FireSimRocketConfig-BaseVitisConfig
ERROR: A bitstream build failed.
```

ncppd, 2023-01-26:
I get the same error, it seems like timing constraints are not propagating correctly in the design.

abejgonzalez, 2023-01-26:
Have you all tried running with 2022.1?

ncppd, 2023-01-26:
I am using Vivado 2022.2, should I switch to 2022.1?

abejgonzalez, 2023-01-26:
I realized I misspoke and I was able to run things successfully on 2022.1. If you have time, I would try that in the meantime.

caizixian, 2023-01-27:
2022.1 builds fine. Thanks a lot for the help.

closed 2023-01-27 by caizixian