firesim/firesim #1305: TracerV-based Targets using BaseVitisConfig

davidbiancolin, 2022-11-18:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

main (11/17/2022)
SHA: 4cd74f89 

### OS Setup

firesim1, azure instances with vitis 2022.1

### Other Setup

### Current Behavior

FPGA deployements and metasims deadlock at cycle 2.

### Expected Behavior

They should simulate.

### Other Information

I dug into this a bit and it seems like the token channels for triggerCredits and triggerDebits are not being driven. This suggests to me that something about the assignment to these channels from the bridges has broken, likely as a side effect of changing some channel handling logic in SimWrapper.

davidbiancolin, 2022-11-18:
@nandor FYI

PKUZHOU, 2022-12-15:
Have you found a solution? I am also faced with this problem...

abejgonzalez, 2022-12-22:
I can confirm this is also happening on a non-TracerV target: https://github.com/firesim/firesim/actions/runs/3755014646/jobs/6379727415

davidbiancolin, 2022-12-22:
Are you certain it's non-tracerv? If so can you provide the cause?

PKUZHOU, 2022-12-22:
By using metasim, I have found that the TracerVBridge has broken.  The TracerV's channel always has out_ready = 0 so that the channel gets stuck at cycle 2:
[image]
In the generated verilog file, I can see that this is because the io_out_ready signal is constantly set to zero. 
[image]

I can also find that some ports of TracerVBridge are missing:
[image]

Here is a functional version derived from https://github.com/EECS-NTNU/firesim/tree/1.15.1_u250

[image]

Do you have any ideas about this? It looks like TracerVBridge is not correctly connected into the simulator during the compiling phase.

davidbiancolin, 2022-12-22:
Yes, this is consistent with the initial bug report. @PKUZHOU what version of FireSim are you using?

PKUZHOU, 2022-12-22:
Hi @davidbiancolin, I have tested both 1.15.1 and  the main branch, they have the same problem. By the way, I have also tested the micro2022_tutorial branch but it also got stuck at Cycle 2

davidbiancolin, 2022-12-22:
Ok thanks.

davidbiancolin, 2022-12-22:
Ok thanks.

PKUZHOU, 2022-12-22:
By removing WithTracerVBridge from the config, firesim works well on my U200 FPGA. Branch 1.15.1

davidbiancolin, 2022-12-24:
Recent changes (went in yesterday) i suspect resolved this indirectly. 

The route cause if i had to guess was that because there was no PCIM, GG/FIRRTL optimized away all of the tracerV logic and tied off the channels.

PKUZHOU, 2023-02-16:
Hi, David

May I ask if you have solved this issue?