firesim/firesim #1420: U250 runworkload aborts

ncppd, 2023-01-31:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

1.15.1-196-g0c2e8dd6

### OS Setup

```
<!-- copy/paste the output of `uname -a; lsb_release -a` below -->
```
Linux quest 4.15.0-202-generic #213-Ubuntu SMP Thu Jan 5 19:19:12 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic

### Other Setup

I have successfully completed `buildbitstream` and `infrasetup` steps, and have untared `driver-bundle.tar.gz` in the simulation directory.

### Current Behavior

When i try to run `firesim runworkload`, simulation stops with the following uart log:

> Script started on 2023-01-31 19:24:24+0200
DEBUG: DevIdx:0 XCLBin:/local/ncpapad/firesim/bitstream//platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/build_dir.xilinx_u250_gen3x16_xdma_4_1_202210_1/firesim.xclbin

> FPGA-managed streams are not yet supported
> bash: line 1:  8608 Aborted                 
> sudo ./FireSim-vitis +permissive +mm_relaxFunctionalModel_0=0 +mm_writeMaxReqs_0=10 +mm_readMaxReqs_0=10 +mm_writeLatency_0=30 +mm_readLatency_0=30 +slotid=0 +binary_file=/local/ncpapad/firesim/bitstream//platforms/vitis/cl_FireSim-FireSimRocketConfig-BaseVitisConfig/bitstream/build_dir.xilinx_u250_gen3x16_xdma_4_1_202210_1/firesim.xclbin +profile-interval=-1 +macaddr0=00:12:6D:00:00:02 +blkdev0=linux-uniform0-br-base.img +niclog0=niclog0 +blkdev-log0=blkdev-log0 +trace-select=1 +trace-start=0 +trace-end=-1 +trace-output-format=0 +dwarf-file-name=linux-uniform0-br-base-bin-dwarf +autocounter-readrate=0 +autocounter-filename-base=AUTOCOUNTERFILE +drj_dtb=linux-uniform0-br-base-bin.dtb +drj_bin=linux-uniform0-br-base-bin +drj_rom=linux-uniform0-br-base-bin.rom +print-start=0 +print-end=-1 +linklatency0=6405 +netbw0=200 +shmemportname0=default +permissive-off +prog0=linux-uniform0-br-base-bin
> Script done on 2023-01-31 19:24:28+0200

It seems that function `char *simif_vitis_t::get_memory_base() ` in `sim/midas/src/main/cc/simif_vitis.cc` is aborting the simulation. This function is called by `FPGAManagedStreamWidget::FPGAManagedStreamWidget` (file `sim/midas/src/main/cc/bridges/fpga_managed_stream.cc`).  In my understanding, since `get_memory_base` is not implemented then simulation cannot commence, right?

### Expected Behavior

Simulation should not abort.

whats4lunch, 2023-02-08:
You should modify the hard coded u250-based address of memory in sim/midas/src/main/cc/simif_vitis.cc

Use xclbinutil to check your appropriate u200 base address

ncppd, 2023-03-09:
By modifying the hardcoded address firesim works as expected. @abejgonzalez is this a workaround? There is no such step in the docs.

davidbiancolin, 2023-03-09:
Can you report the address you had to change it to?

The right solution here is to invest the time to allow the driver to pass in the base, which can vary across shell versions and HW_EMU.

ncppd, 2023-03-11:
That makes sense, in my case the base address is `0xc10000`. I am using platform version `xilinx_u250_gen3x16_xdma_4_1_202210_1`.

closed 2025-05-02 by ncppd