firesim/firesim #1455: mismatching rsync versions on manager and remote hosts causes issues

timsnyder-siv, 2023-03-06:
I've recently run into issues where I see rsync fail with:
```
2023-03-06--03-40-51-infrasetup-VXMYBC69ALNO3NYZ.log:2023-03-06 03:41:43,238 [flush       ] [INFO ]  Fatal error: local() encountered an error (return code 12) while executing 'rsync  -pthrvz -L --rsh='ssh   -o StrictHostKeyChecking=no ' workloads/spec2k6-all-p670-thp-0x8224AE0C3E241_0x929f-vect-ref-sifive_p670_clang_static/../../../firemarshal-sifive/boards/firechip/base-workloads/fusdk-p670/fusdk-p670.img centos@tim-run49-vm:/home/centos/sim_slot_0/rsyncdir/fusdk-p670.img'
2023-03-06--03-40-51-infrasetup-VXMYBC69ALNO3NYZ.log-2023-03-06 03:41:43,238 [flush       ] [INFO ]  =============================== Standard output ===============================
2023-03-06--03-40-51-infrasetup-VXMYBC69ALNO3NYZ.log-2023-03-06 03:41:43,238 [flush       ] [INFO ]  sending incremental file list
2023-03-06--03-40-51-infrasetup-VXMYBC69ALNO3NYZ.log-2023-03-06 03:41:43,238 [flush       ] [INFO ]  fusdk-p670.img
2023-03-06--03-40-51-infrasetup-VXMYBC69ALNO3NYZ.log-2023-03-06 03:41:43,238 [flush       ] [INFO ]  ================================================================================
2023-03-06--03-40-51-infrasetup-VXMYBC69ALNO3NYZ.log-2023-03-06 03:41:43,238 [flush       ] [INFO ]  ================================ Standard error ================================
2023-03-06--03-40-51-infrasetup-VXMYBC69ALNO3NYZ.log-2023-03-06 03:41:43,238 [flush       ] [INFO ]  deflate on token returned 0 (11241 bytes left)
2023-03-06--03-40-51-infrasetup-VXMYBC69ALNO3NYZ.log-2023-03-06 03:41:43,238 [flush       ] [INFO ]  rsync error: error in rsync protocol data stream (code 12) at token.c(481) [sender=3.2.7]
2023-03-06--03-40-51-infrasetup-VXMYBC69ALNO3NYZ.log-2023-03-06 03:41:43,238 [flush       ] [INFO ]  ================================================================================
2023-03-06--03-40-51-infrasetup-VXMYBC69ALNO3NYZ.log-2023-03-06 03:41:43,238 [flush       ] [INFO ]  Aborting.
```

I google searched and determined it was being caused by different versions of rsync being installed on my manager and runhosts.  On the manager, we're installing `rsync` with conda but leaving the runhosts to using whatever system rsync is installed.

I was able to work around this by removing `rsync` from my conda environment but we should probably have a better solution.

timsnyder-siv, 2023-03-06:
To make the error message earlier to read:
```
rsync error: error in rsync protocol data stream (code 12) at token.c(481) [sender=3.2.7]
```

SeongRyong0726, 2024-06-04:
I also face this issue now.
Did you solve this problem?