firesim/firesim #1686: NoMemPortConfig broken due to no `loadmem.h` present

abejgonzalez, 2024-01-08:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

Current `main`

### OS Setup

N/A

### Other Setup

### Current Behavior

NoMemPortConfig fails due to not finding `loadmem.h` in the build (normally it is included by default).

### Expected Behavior

NoMemPortConfig should build.

### Other Information

This is the fix
```
diff --git a/sim/midas/src/main/cc/entry.cc b/sim/midas/src/main/cc/entry.cc
index d1766090a..5f6667397 100644
--- a/sim/midas/src/main/cc/entry.cc
+++ b/sim/midas/src/main/cc/entry.cc
@@ -10,6 +10,7 @@
 #define GET_INCLUDES
 #include "FireSim-generated.const.h"
 #undef GET_INCLUDES
+#include "bridges/loadmem.h" // always ensure that loadmem.h header is present for tsibridge to use loadmem

 #define GET_METASIM_INTERFACE_CONFIG
 #define GET_SUBSTRUCT_CHECKS
 ```

abejgonzalez, 2024-01-09:
```
template <typename T>
  T *get_widget_opt() {
    auto it = widgets.find(&T::KIND);
    if (it == widgets.end() || it->second.size() != 1)
      return nullptr;
    return static_cast<T *>(it->second[0].get());
  }
```
  
  Probably need to use this too.

muhammadali201, 2025-01-10:
Whats the point of loadmem when you have built NoMemPortConfig?
When I look at the generated header fo NoMemPort Config, I don't see FASEDMemoryMode and LOADMEMWidget which is obvious as we are building a NoMemPort config and, loadmem widget is not added to widget registery. 

Question:
Why does constructor to TSIBridge needs to get loadmem_t widget from widget registery, especially when You have built a config without loadmem widget?
What is the purpose of Loadmem in NoMemPort Config?

This also causes confusion when adding support for new FPGA card. Firesim latest docs advise that one should start with Just AXI-Lite (MMIO) port and than later on add DMA / DRAM ports. The apropriate config for this would be NoMemPortConfig with MMIO only bridges.
However, NoMemPort config crashes during simulation because of TSI Bridge. This means that for new FPGA card, one should start with MMIO and DRAM ports (instead of starting with Just MMIO port)

abejgonzalez, 2025-01-14:
You are correct, we shouldn't need the loadmem widget for a `NoMemPortConfig`. However, the way the current TSI Bridge is implemented you need a reference to the widget code even if the widget isn't used.