firesim/firesim #1688: Exit quickly without uartlog when running 'linux-uniform' workload with a single rocket on U280

SeHuibixiaoxing, 2024-01-20:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

1.17.1-0-gb473147af

### OS Setup

Linux c111 5.4.0-150-generic #167~18.04.1-Ubuntu SMP Wed May 24 00:51:42 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
LSB Version:	core-9.20170808ubuntu1-noarch:printing-9.20170808ubuntu1-noarch:security-9.20170808ubuntu1-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic

### Other Setup

Follow the [Xilinx Alveo U280 XDMA-based Getting Started Guide](https://docs.fires.im/en/1.17.1/Getting-Started-Guides/On-Premises-FPGA-Getting-Started/Xilinx-Alveo-U280-FPGAs.html) to complete the configuration

### Current Behavior

**Minimum example**
I clone and config new firesim 1.17.1, download  `alveo_u280_firesim_rocket_singlecore_no_nic` pre-build bitstream, compile `linux-uniform.json`. There are no errors reported during `firesim enumeratefpgas` or `firesim runworkload`. After running `firesim runworkload`, the program quickly exits. 

**Background(Perhaps it can be ignored)**
At the beginning, I was trying to replicate an open-source work on U280. Quickly exit when executing `firesim runworkload`. There is only one workload folder in the workload results directory (there are actually two workloads), and there is no uartlog file. There is no error message in the log file.
Afterwards, I ran workload with `alveo_u280_firesim_rocket_singlecore_no_nic` and `linux-uniform.json` in the same firesim workspace. It still exit quickly without uartlog files.

### Expected Behavior

 I have successfully run `alveo_u280_firesim_rocket_singlecore_no_nic` and `linux-uniform.json` before. But it seems that it's not working now. I haven't made any other modifications. If a new project is cloned and completed according to the configuration process, it can run successfully because it has been successful before.

### Other Information

deploy/results-workload/2024-01-19--17-41-55-linux-uniform/linux-uniform0/HW_CFG_SUMMARY
```
RuntimeHWConfig: alveo_u280_firesim_rocket_singlecore_no_nic
DeployQuintuplet: xilinx_alveo_u280-firesim-FireSim-FireSimRocketConfig-BaseXilinxAlveoConfig
AGFI: None
Bitstream tar: file:///mnt/eda2/wangzy/Downloads/alveo_u280_firesim_rocket_singlecore_no_nic.tar.gz
CustomRuntimeConf: None
```
deploy/results-workload/2024-01-19--17-41-55-linux-uniform/linux-uniform0/os-release
```
NAME=Buildroot
VERSION=2022.11.1
ID=buildroot
VERSION_ID=2022.11.1
PRETTY_NAME="Buildroot 2022.11.1"
```
deploy/results-workload/2024-01-19--17-41-55-linux-uniform/linux-uniform0/sim-run.sh
```
#!/usr/bin/env bash
screen -S fsim0 -d -m bash -c "script -f -c 'stty intr ^] && sudo ./FireSim-xilinx_alveo_u280 +permissive   +macaddr0=00:12:6D:00:00:02 +blkdev0=linux-uniform0-br-base.img +niclog0=niclog0 +blkdev-log0=blkdev-log0  +trace-select=1 +trace-start=0 +trace-end=-1 +trace-output-format=0 +dwarf-file-name=linux-uniform0-br-base-bin-dwarf +autocounter-readrate=0 +autocounter-filename-base=AUTOCOUNTERFILE +drj_dtb=linux-uniform0-br-base-bin.dtb +drj_bin=linux-uniform0-br-base-bin +drj_rom=linux-uniform0-br-base-bin.rom  +print-start=0 +print-end=-1 +linklatency0=6405 +netbw0=200 +shmemportname0=default  +domain=0x0000 +bus=0xd8 +device=0x00 +function=0x0 +bar=0x0 +pci-vendor=0x10ee +pci-device=0x903f +permissive-off +prog0=linux-uniform0-br-base-bin   && stty intr ^c' uartlog"; sleep 1
```

`runworkload` log file:
```
2024-01-19 12:41:55,823 [main        ] [INFO ]  FireSim Manager. Docs: https://docs.fires.im
Running: runworkload

2024-01-19 12:41:55,830 [__init__    ] [DEBUG]  {'config_file_name': 'config_hwdb.yaml',
 'hwconf_dict': {'alveo_u200_firesim_rocket_singlecore_no_nic': <runtools.runtime_config.RuntimeHWConfig object at 0x7f0a1ae849d0>,
                 'alveo_u250_firesim_gemmini_rocket_singlecore_no_nic': <runtools.runtime_config.RuntimeHWConfig object at 0x7f0a1ae84a60>,
                 'alveo_u250_firesim_rocket_singlecore_no_nic': <runtools.runtime_config.RuntimeHWConfig object at 0x7f0a1ae84970>,
                 'alveo_u280_firesim_rocket_singlecore_no_nic': <runtools.runtime_config.RuntimeHWConfig object at 0x7f0a1ae84b80>,
[... 511 lines cut ...]

sent 9.21K bytes  received 172 bytes  6.26K bytes/sec
total size is 28.66K  speedup is 3.05
2024-01-19 12:40:58,458 [flash_fpgas ] [DEBUG]  
2024-01-19 12:40:58,459 [instance_log] [INFO ]  [localhost] Determine BDF for 0
2024-01-19 12:40:58,459 [flush       ] [DEBUG]  [localhost] run: cat /opt/firesim-db.json
2024-01-19 12:40:58,798 [flush       ] [DEBUG]  [localhost] out: [
2024-01-19 12:40:58,798 [flush       ] [DEBUG]  [localhost] out:   {
2024-01-19 12:40:58,799 [flush       ] [DEBUG]  [localhost] out:     "uid": "Xilinx/217702074014A",
2024-01-19 12:40:58,799 [flush       ] [DEBUG]  [localhost] out:     "device": "xcu280_u55c_0",
2024-01-19 12:40:58,799 [flush       ] [DEBUG]  [localhost] out:     "bdf": "d8:00.0"
2024-01-19 12:40:58,800 [flush       ] [DEBUG]  [localhost] out:   }
2024-01-19 12:40:58,800 [flush       ] [DEBUG]  [localhost] out: ]
2024-01-19 12:40:58,801 [instance_log] [INFO ]  [localhost] Flashing FPGA Slot: 0 (d8:00.0) with bitstream: /mnt/eda2/wangzy/proj/groundtruth/firesim/FIRESIM_RUNS_DIR/sim_slot_0//xilinx_alveo_u280/firesim.bit
2024-01-19 12:40:58,801 [flush       ] [DEBUG]  [localhost] run: /mnt/eda2/wangzy/proj/groundtruth/firesim/FIRESIM_RUNS_DIR/sim_slot_0//scripts/fpga-util.py --bitstream /mnt/eda2/wangzy/proj/groundtruth/firesim/FIRESIM_RUNS_DIR/sim_slot_0//xilinx_alveo_u280/firesim.bit --bdf d8:00.0
2024-01-19 12:40:59,221 [flush       ] [DEBUG]  [localhost] out: :INFO: Running: ['/usr/bin/sudo', '/mnt/eda2/wangzy/proj/groundtruth/firesim/FIRESIM_RUNS_DIR/sim_slot_0/scripts/fpga-util.py', '--bitstream', '/mnt/eda2/wangzy/proj/groundtruth/firesim/FIRESIM_RUNS_DIR/sim_slot_0//xilinx_alveo_u280/firesim.bit', '--bdf', 'd8:00.0', '--vivado-bin', '/mnt/eda1/Vivado/2021.1/bin/vivado', '--hw-server-bin', '/mnt/eda1/Vivado/2021.1/bin/hw_server']
2024-01-19 12:40:59,336 [flush       ] [DEBUG]  [localhost] out: :INFO: Writing to /sys/devices/pci0000:d7/0000:d7:00.0/0000:d8:00.0/remove: 1
2024-01-19 12:41:33,479 [flush       ] [DEBUG]  [localhost] out: :INFO: Successfully programmed FPGA d8 with /mnt/eda2/wangzy/proj/groundtruth/firesim/FIRESIM_RUNS_DIR/sim_slot_0/xilinx_alveo_u280/firesim.bit
2024-01-19 12:41:33,483 [flush       ] [DEBUG]  [localhost] out: :INFO: Writing to /sys/bus/pci/rescan: 1
2024-01-19 12:41:33,559 [flush       ] [DEBUG]  [localhost] out: 
2024-01-19 12:41:33,568 [flush       ] [DEBUG]  [localhost] run: lsmod | grep -wq xdma
2024-01-19 12:41:33,919 [instance_log] [INFO ]  [localhost] XDMA Driver Kernel Module already loaded.
2024-01-19 12:41:33,973 [<module>    ] [INFO ]  The full log of this run is:
/mnt/eda2/wangzy/proj/groundtruth/firesim/deploy/logs/2024-01-19--17-40-31-infrasetup-MJG32SGCI4YSF26O.log
```

Han-sok, 2024-06-10:
I am experiencing exactly the same problem as you stated. Did you find out what was the cause and solve the problem?
Perhaps, the line in the run log,
"2024-01-19 12:41:59,569 [flush       ] [DEBUG]  [localhost] out: ./sim-run.sh: line 2: screen: command not found"
was the culprit?

Let me know what was your solution.

SeHuibixiaoxing, 2024-06-10:
> I am experiencing exactly the same problem as you stated. Did you find out what was the cause and solve the problem? Perhaps, the line in the run log, "2024-01-19 12:41:59,569 [flush ] [DEBUG] [...]

连接后发现没有screen。我应该是首先通过"ssh localhost"连接到run machine.而后使用apt安装了screen。 
114/10000实时翻译 
 
It's been a bit long. At that time, I had already installed the screen locally. But for some reason, after connecting through SSH localhost, I found that there was no screen. I should have first connected to the run machine through "ssh localhost" and then installed the screen using apt.