firesim/firesim #1692: Linux doesn't boot with default config on latest v1.18.0 release

jaisw7, 2024-01-27:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

```
1.18.0-0-g535dcdc29
```

### OS Setup

```
Linux ip-192-168-1-182.ec2.internal 3.10.0-1160.81.1.el7.x86_64 #1 SMP Fri Dec 16 17:29:43 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
LSB Version:    :core-4.1-amd64:core-4.1-noarch
Distributor ID: CentOS
Description:    CentOS Linux release 7.9.2009 (Core)
Release:        7.9.2009
Codename:       Core
```

### Other Setup

### Current Behavior

```
Script started, file is uartlog
AFI ID for Slot  0: agfi-02769e6a3f322ee10
AFI PCI  Vendor ID: 0x1d0f, Device ID 0xf000
Using xdma write queue: /dev/xdma0_h2c_0
Using xdma read queue: /dev/xdma0_c2h_0
UART0 is here (stdin/stdout).
TraceRV 0: Tracing disabled, since +tracefile was not provided.
TraceRV 1: Tracing disabled, since +tracefile was not provided.
TraceRV 2: Tracing disabled, since +tracefile was not provided.
TraceRV 3: Tracing disabled, since +tracefile was not provided.
command line for program 0. argc=20:
+permissive +macaddr0=00:12:6D:00:00:02 +blkdev0=linux-uniform0-br-base.img +niclog0=niclog0 +blkdev-log0=blkdev-log0 +trace-select=1 +trace-start=0 +trace-end=-1 +trace-output-format=0 +dwarf-file-name=linux-uniform0-br-base-bin-dwarf +autocounter-readrate=0 +autocounter-filename-base=AUTOCOUNTERFILE +print-start=0 +print-end=-1 +linklatency0=6405 +netbw0=200 +shmemportname0=default +slotid=0 +permissive-off linux-uniform0-br-base-bin
FireSim fingerprint: 0x46697265
TracerV: Trigger enabled from 0 to 18446744073709551615 cycles
TracerV: Trigger enabled from 0 to 18446744073709551615 cycles
TracerV: Trigger enabled from 0 to 18446744073709551615 cycles
TracerV: Trigger enabled from 0 to 18446744073709551615 cycles
Commencing simulation.
```

Hangs forever

### Expected Behavior

Following the standard firesim docs, with the default configuration, a user should be able to boot linux.

RealJustinNi, 2024-02-27:
Same problem.

joonho3020, 2024-02-27:
Which config are you using?

SwiftCrown, 2024-02-27:
> Which config are you using? Do you have a NIC in your design?

No, we use quadcore with uniform-Linux on U200 card and rerocc with multi-Gemmini and uniform-Linux on U200 card.

gonsolo, 2024-03-16:
I have the same problem. @abejgonzalez

suhwanhong999, 2024-11-30:
Was anyone able to solve this issue? I am having the same exact issue where the simulation hangs at 'commencing simulation', at least for ~20 mins where I forced a stop.