firesim/firesim #1793: Support for local FPGA board

cmx-Y, 2024-08-12:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### Feature Description

In addition to the AWS F1, U200, U280 and other boards officially supported by Firesim, what other boards does Firesim support, such as whether it supports VC707? Is there a tutorial to help users run Firesim on a local FPGA board?

### Motivating Example

We want to get the specific performance of the workload running on Gemmini, so we need to use the Firesim project. However, we only have the VC707 FPGA board, and Firesim does not directly support this board, so we encountered some difficulties in running Firesim. In the mailing list, we saw that David successfully ran Firesim on the local U250, and we also saw Yang Wang trying to run Firesim on the XCVU13P board, but many specific details were not given. We would like to know what are the general processes required to adapt Firesim to a specific FPGA board? What files in Firesim do we need to edit?

muhammadali201, 2024-09-07:
I think, for an experience firesim use, it would take 3-4 weeks to support VC707. I think the biggest issue will be XDMA RTL and driver for VC707.

muhammadali201, 2024-09-17:
Actually, I think it should take less than 3 weeks.

Start with firesim docs [this](https://docs.fires.im/en/latest/Advanced-Usage/Adding-FPGAs.html) section. 

Than you have to tweak the settings of xdma ip for PCIe Gen2 devices. Not sure How many PCIe lanes does VC707 has. Also there are some features in xdma ip that are only applicable to US+ devices. 

You will also have to change the MIG IP to generate DDR3 controller. 

Some changes maybe needed in Fased memory model to support DDR3 devices

zhanghb97, 2024-09-18:
Thank you very much for your detailed response! We will follow the process you outlined and start making the necessary adjustments. If we encounter any issues, we’ll reach out for further guidance. Appreciate your help!

muhammadali201, 2024-09-18:
Just out of curiosity, what exactly are you trying to do? VC707 FPGA is relatively small device, can only accodomate small designs.

cmx-Y, 2024-09-19:
We want to refine the existing Gemmini LLVM backend infrastructure and construct a robust Just-In-Time (JIT) execution engine in buddy-mlir repository. The engine is designed to effectively interpret and execute LLVM IR or dialects within the Gemmini simulator/evaluation platform, optimizing for high performance and efficiency. However, the Spike simulator provided by Gemmini can only verify functional correctness; it cannot give accurate performance metrics or profiling information. And the cycle-accurate simulator Verilator runs slowly. So we want to use Firesim to speed up the simulation. If you are interested in our ongoing work, you can check out this website for more details: https://buddy-compiler.github.io/.