firesim/firesim #1842: Issue with Setting up AWS F1 and Setting up the FireSim Repo

arunlee77, 2024-11-28:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

### OS Setup

```
[centos@ip-172-31-25-181 ~]$ /etc/os-release
-bash: /etc/os-release: Permission denied
[centos@ip-172-31-25-181 ~]$ cat /etc/os-release
NAME="CentOS Linux"
VERSION="7 (Core)"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="7"
PRETTY_NAME="CentOS Linux 7 (Core)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:7"
HOME_URL="https://www.centos.org/"
BUG_REPORT_URL="https://bugs.centos.org/"

CENTOS_MANTISBT_PROJECT="CentOS-7"
CENTOS_MANTISBT_PROJECT_VERSION="7"
REDHAT_SUPPORT_PRODUCT="centos"
REDHAT_SUPPORT_PRODUCT_VERSION="7"

```

### Other Setup

https://docs.fires.im/en/latest/Getting-Started-Guides/AWS-EC2-F1-Getting-Started/Setting-Up-The-FireSim-Repo.html

### Current Behavior

I am trying to set up the firsim on AWS F1 following  https://docs.fires.im/en/latest/Getting-Started-Guides/AWS-EC2-F1-Getting-Started/Setting-Up-The-FireSim-Repo.html

Running ./build-setup.sh gives me the following error

`
INFO:root:  xz                 conda-forge/linux-64::xz-5.2.6-h166bdaf_0
INFO:root:  yaml               conda-forge/linux-64::yaml-0.2.5-h7f98852_2
INFO:root:  yarl               conda-forge/linux-64::yarl-1.9.4-py310h2372a71_0
INFO:root:  zipp               conda-forge/noarch::zipp-3.19.2-pyhd8ed1ab_0
INFO:root:  zlib               conda-forge/linux-64::zlib-1.2.13-h4ab18f5_6
INFO:root:  zstandard          conda-forge/linux-64::zstandard-0.23.0-py310h64cae3c_0
INFO:root:  zstd               conda-forge/linux-64::zstd-1.5.6-ha6fb4c9_0
INFO:root:
INFO:root:
INFO:root:
INFO:root:Downloading and Extracting Packages: ...working... done
INFO:root:Preparing transaction: done
INFO:root:Verifying transaction: done
INFO:root:Executing transaction: -
INFO:roo-
INFO:roodone
INFO:root:#
INFO:root:# To activate this environment, use
INFO:root:#
INFO:root:#     $ conda activate /home/centos/chipyard/.conda-env
INFO:root:#
INFO:root:# To deactivate an active environment, use
INFO:root:#
INFO:root:#     $ conda deactivate
INFO:root:
ERROR:root:ERROR: bcrypt-4.2.0-cp39-abi3-manylinux_2_28_x86_64.whl is not a supported wheel on this platform.
ERROR:root:
ERROR:root:ERROR conda.cli.main_run:execute(125): `conda run pip install --no-deps -r /tmp/tmprc2g8m5u` failed. (See above for error)
Traceback (most recent call last):
  File "/home/centos/chipyard/.conda-lock-env/bin/conda-lock", line 10, in <module>
    sys.exit(main())
             ~~~~^^
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/conda_lock/conda_lock.py", line 1498, in click_install
    install(
    ~~~~~~~^
        conda=conda,
        ^^^^^^^^^^^^
    ...<10 lines>...
        extras=extras,
        ^^^^^^^^^^^^^^
    )
    ^
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/conda_lock/conda_lock.py", line 1552, in install
    install_func(file=lockfile)
    ~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/conda_lock/conda_lock.py", line 240, in do_conda_install
    _conda(["run"], ["pip", "install", "--no-deps", "-r", str(requirements_path)])
    ~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/conda_lock/invoke_conda.py", line 143, in _invoke_conda
    raise subprocess.CalledProcessError(
    ...<4 lines>...
    )
conda_lock._vendor.poetry.utils._compat.CalledProcessError: Command '['/opt/conda/condabin/conda', 'run', '--prefix', '/home/centos/chipyard/.conda-env', 'pip', 'install', '--no-deps', '-r', '/tmp/tmprc2g8m5u']' returned non-zero exit status 1.
build-setup.sh: Build script failed with exit code 1 at step 1: Conda environment setup
`

AWS F1 instance was setup following 

https://docs.fires.im/en/latest/AWS-EC2-F1-Initial-Setup/Setting-up-your-Manager-Instance.html

### Expected Behavior

Able to setup Firesim without errors

### Other Information

Tried with a more recent version of chipyard, but the issue still persists.

arunlee77, 2024-11-28:
The error is 

```

ERROR:root:ERROR: bcrypt-4.2.0-cp39-abi3-manylinux_2_28_x86_64.whl is not a supported wheel on this platform.
ERROR:root:
ERROR:root:ERROR conda.cli.main_run:execute(125): `conda run pip install --no-deps -r /tmp/tmp3dpq11q3` failed. (See above for error)
Traceback (most recent call last):
  File "/home/centos/chipyard/.conda-lock-env/bin/conda-lock", line 10, in <module>
    sys.exit(main())
             ~~~~^^
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/conda_lock/conda_lock.py", line 1498, in click_install
    install(
    ~~~~~~~^
        conda=conda,
        ^^^^^^^^^^^^
    ...<10 lines>...
        extras=extras,
        ^^^^^^^^^^^^^^
    )
    ^
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/conda_lock/conda_lock.py", line 1552, in install
    install_func(file=lockfile)
    ~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/conda_lock/conda_lock.py", line 240, in do_conda_install
    _conda(["run"], ["pip", "install", "--no-deps", "-r", str(requirements_path)])
    ~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/centos/chipyard/.conda-lock-env/lib/python3.13/site-packages/conda_lock/invoke_conda.py", line 143, in _invoke_conda
    raise subprocess.CalledProcessError(
    ...<4 lines>...
    )
conda_lock._vendor.poetry.utils._compat.CalledProcessError: Command '['/opt/conda/condabin/conda', 'run', '--prefix', '/home/centos/chipyard/.conda-env', 'pip', 'install', '--no-deps', '-r', '/tmp/tmp3dpq11q3']' returned non-zero exit status 1.
build-setup.sh: Build script failed with exit code 1 at step 1: Conda environment setup
[centos@ip-172-31-23-134 chipyard]$
```

suhwanhong999, 2024-11-30:
Did you install conda-lock version 1.4.0? If you did, did you run 'conda activate base'? I know this technically applies to the chipyard repository (anything before the latest 1.13.0 version) but this seems very similar to the bug I had.

arunlee77, 2024-12-02:
As per the firesim documentation, the setup follows a different path than mentioned in chipyard.
https://docs.fires.im/en/latest/AWS-EC2-F1-Initial-Setup/Setting-up-your-Manager-Instance.html

Not sure whether the chipyard doc to be followed or not!!

arunlee77, 2024-12-02:
I tried following the chipyard documentation after setting up the manager instance and i get the same error.

```
ERROR:root:ERROR: bcrypt-4.2.0-cp39-abi3-manylinux_2_28_x86_64.whl is not a supported wheel on this platform.
ERROR:root:
ERROR:root:ERROR conda.cli.main_run:execute(125): `conda run pip install --no-deps -r /tmp/tmptg7yfyer` failed. (See above for error)

```

so both chipyard docus and firesim doc are not working for me with the latest centos FPGA Developer AMI image.

suhwanhong999, 2024-12-02:
If you are using the latest user data setup script, I haven't had any issues for the number of instances I've made. Have you tried just deleting the instance and recreating it?

arunlee77, 2024-12-03:
What i get from AWS is the following `FPGA Developer AMI - 1.12.2-40257ab5-6688-4c95-97d1-e251a40fd1fc ` and the script i am using is from https://docs.fires.im/en/latest/AWS-EC2-F1-Initial-Setup/Setting-up-your-Manager-Instance.html .
I have done re creating the instance a couple of times already. will try again.

arunlee77, 2024-12-03:
A new failure now:

```
make[1]: Leaving directory `/usr/src/kernels/3.10.0-1160.81.1.el7.x86_64'
Loaded plugins: fastestmirror
Determining fastest mirrors
Could not retrieve mirrorlist http://mirrorlist.centos.org/?release=7&arch=x86_64&repo=os&infra=genclo error was
14: curl#6 - "Could not resolve host: mirrorlist.centos.org; Unknown error"

 One of the configured repositories failed (Unknown),
 and yum doesn't have enough cached data to continue. At this point the only
 safe thing yum can do is fail. There are a few ways to work "fix" this:

```

arunlee77, 2024-12-03:
This issue can be fixed with baseurl=http://vault.centos.org change in the repo configuration.

closed 2024-12-05 by arunlee77