firesim/firesim #1848: Midas cannot find CDE

KarlJoad, 2024-12-23:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

```console
chipyard/sims/firesim $ git describe --tags --long
1.20.1-51-g141bff735
```

This was done with the documentation on "Latest", which says to clone Chipyard and have Chipyard initialize Firesim as a simulator submodule.
```console
chipyard $ git describe --tags --long
1.13.0-0-g69eba860
```

```console
chipyard $ git log 
commit 69eba860a352343e4ac6b6df0f3638a79a86ec78 (HEAD, tag: 1.13.0)
```

### OS Setup

```console
$ uname -a; lsb_release -a
Linux dubliner 5.15.0-125-generic #135-Ubuntu SMP Fri Sep 27 13:53:58 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.4 LTS
Release:        22.04
Codename:       jammy
```

### Other Setup

https://docs.fires.im/en/latest/Getting-Started-Guides/On-Premises-FPGA-Getting-Started/Repo-Setup/Xilinx-VCU118.html

As stated in first code-block, `main` of Chipyard was used, instead of the commit hash used in the documentation.

**NOT using Conda!**
I have developed a Nix flake that provides all system dependencies that are required. I compile the GCC toolchain & sysroot, and LLVM/CIRCT manually.

### Current Behavior

After several script & environment variable issues, I got through `managerinit` but `enumeratefpgas` fails.

```console
$ firesim enumeratefpgas
[localhost] out: [error] /pool/karl/chipyard-firesim/sims/firesim/sim/midas/src/main/scala/configs/CompilerConfigs.scala:5:12: object chipsalliance is not a member of package org
[localhost] out: [error] import org.chipsalliance.cde.config.{Config, Parameters}
[localhost] out: [error]            ^
```

### Expected Behavior

Midas should find CDE in Scala/Java's import path, allowing Midas to compile.

### Other Information

[2024-12-23--22-07-21-enumeratefpgas-7CZR1YQ6K61ZLULO.log](https://github.com/user-attachments/files/18233895/2024-12-23--22-07-21-enumeratefpgas-7CZR1YQ6K61ZLULO.log)

tanneberger, 2025-04-18:
@KarlJoad Out of curiosity did you find a solution/workaround to this problem?

KarlJoad, 2025-04-29:
Sorry for the delay @tanneberger. I have since deleted the version of Chipyard/Firesim that had this issue. I just tried reproducing it with Chipyard's current `main` (`0c91ddb1186c1b8c13fdb54bce6fa6f8e1d5cbe7`) and could not make it happen.

I used the same `flake.nix` as when I opened this issue (with minor additions for the additions made to Firesim) and got past this error. Seems that it was either fixed or that I did things in the proper order this time? I honestly don't know which/what.

But I end up in the same state as #1849, where `enumeratefpgas` cannot proceed further because of bad/missing environment variable exports.