firesim/firesim #1850: Default RocketChip `buildbitstream` for Xilinx VCU118 fails on 1.20.1

egelja, 2025-01-12:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/firesim)
- [X] Yes, I searched [prior issues](https://github.com/firesim/firesim/issues)
- [X] Yes, I searched the [documentation](https://docs.fires.im/en/stable/)

### FireSim Version and Hash

1.20.1-0-g12e2e0a6b

### OS Setup

```
Linux dubliner 5.15.0-125-generic #135-Ubuntu SMP Fri Sep 27 13:53:58 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.4 LTS
Release:        22.04
Codename:       jammy
```

### Other Setup

I followed these guides from the documentation:

1. https://docs.fires.im/en/1.20.1/Getting-Started-Guides/On-Premises-FPGA-Getting-Started/Repo-Setup/Xilinx-VCU118.html
2. https://docs.fires.im/en/1.20.1/Getting-Started-Guides/On-Premises-FPGA-Getting-Started/Building-a-FireSim-Bitstream/Xilinx-VCU118.html

### Current Behavior

I'm following the guides above, but when I get to the `buildbitstream` step Vivado fails with many errors related to missing ports (see `err.txt`). I have successfully built this on earlier versions on Firesim, so the issue shouldn't be with our FGPA setup.

### Expected Behavior

Build shouldn't fail.

### Other Information

Error log and deployment configs linked below.

[err.txt](https://github.com/user-attachments/files/18390922/err.txt)
[configs.zip](https://github.com/user-attachments/files/18390923/configs.zip)

KarlJoad, 2025-01-22:
You need to pull `platforms/xilinx_vcu118/garnet-firesim` past commit 5bf5b9a2de8c6670024f01334a17da58feb3bdad. That resolves the naming issue that shows up in synthesis.

jimfangx, 2026-10-02:
seems to be resolved, please reopen the issue if that's not the case!

closed 2026-10-02 by jimfangx